adamreeve/npTDMS #21: Version numbering

icare4-rapla, 2015-03-31:
Hi Adam,

Some of my teammembers are using version 0.6.2. But when I compare tdms.py they are all different.
Would it be possible to add a patch level so that we can see immediately the difference?
I suggest to set the version number now to 0.6.3.0000 or so, as a start for synching.

Thanks,

Bob

adamreeve, 2015-04-01:
Hi Bob

Ideally most people would be using the package from PyPI (ie. installing with pip), instead of getting the source code from GitHub, so the same version will always be consistent. Obviously there are cases when someone needs to use a development version from GitHub to get a fix that hasn't been released, so the version might not have been updated, but that should be an exception rather than the default. 

This is probably my fault for making a few changes recently without incrementing the version number and releasing a new PyPI package in a while. I'll try and make releases more frequently when there are changes. I've updated the version to 0.6.3 and uploaded this to PyPI.

Cheers,
Adam

closed 2015-04-01 by adamreeve

icare4-rapla, 2015-04-07:
Hi Adam,

Many thanks.

The quick response restored the confidence of my (civil) engineers.
But the question made sense.
I am sure this product is appreciated by them.

Any new features planned in the near future?

Regards,

Bob

adamreeve, 2015-04-08:
No there's nothing planned by me, I don't work with TDMS files anymore so am just doing maintenance basically. But still happy to fix bugs and accept pull requests.

icare4-rapla, 2015-04-30:
Hi Adam,

Are you using other mx devices than National Instruments now?
Is can you share which one you use and/or the benefits of the change?

Thx,

Bob

adamreeve, 2015-04-30:
No I'm working on web application development now so doing quite different work to my previous research!