adamreeve/npTDMS #22: tdmsfile.group_channels(tdmsfile.groups()[0]) is empty

nathanielatom, 2015-05-25:
Working with a TDMS file which should have multiple groups for decimated versions of data (data with different lengths). Each group should have one or more channels. The command:
``tdmsfile.groups()`` gives:
``[u'22']``
However ``tdmsfile.group_channels(tdmsfile.groups()[0])`` gives an empty list.
The TDMS file was created using NI SignalExpress 2014.
Link to the TDMS file hopefully coming shortly.

adamreeve, 2015-05-26:
Looks like there's a bug with how the group names are handled when they have a "/" in them and are also quoted. I can access the data if I use the correct group names, eg:
```
tdmsfile.group_channels('22/05/2015 04:18:38 PM - Acceleration - Decimated Data_Level1')
```

adamreeve, 2015-05-26:
Should be fixed in commit 3869ce6df, and I've released a bug-fix version 0.6.4 on PyPI.

closed 2015-05-26 by adamreeve

nathanielatom, 2015-05-26:
Updated, and it worked flawlessly. Thank you for your help and ultra-quick response!

adamreeve, 2015-05-26:
No problem, you created the issue at a good time!