adamreeve/npTDMS #24: plans for working on tdsTypeDAQmxRawData?

JuergenNeubauer, 2015-06-10:
Hi,
It seems that NI finally released a full documentation of their TDMS file format, including the DAQmxRawData format: http://www.ni.com/white-paper/5696/en/

Do you have plans to include reading capabilities for this format? It seems the syntax is quite complicated, so a simple hack might not do it?

Thanks, Juergen

adamreeve, 2015-06-12:
Yeah it doesn't look like it would be simple to add support for DAQmxRawData. I don't have any plans to add it myself but if anybody wanted to try and get it working I'd be happy to help out with advice.

JuergenNeubauer, 2015-06-16:
Thanks. After looking more closely at the mentioned documentation, there is no explicit documentation of the data format in the raw data chunks in the DAQmxRawData format. I played a little with pyTDMS and was able to change it to just show me the leadin info of the segments and some raw binary data. I don't know whether I can make more sense of this at this point without a better documentation from NI. That's a very frustrating situation. There seem to be some Matlab scripts out there that empirically hacked some data reading scheme for DAQmxRawData.
I have many files in this particular format where I also have a converted version that adheres to the more standard TDMS data format. So I could potentially experiment with this.

Re-Lien, 2015-11-12:
Dear Juergen and Adam, 

It is really a frustrating situation. 
I would like to acquire and analyse some datas, using python only, and I face this issue with the new TDMS format. 

Is there a workaround that we could use ? 
Adam : it seems that you do not need the support for DAQmxRawData. 
How are you dealing with the TDMS files ? Is there an option in order to choose de 1.0 version of TDMS file ? 

That said, thank you so much for you work. 

Best regards, 

Aurélien

adamreeve, 2015-11-14:
Hi Aurélien

I don't use TDMS files generated by DAQmx, I only work with normal TDMS files, so this isn't an issue for me. DAQmxRawData isn't specific to version 2.0 of the TDMS format, it's just a different type of data segment that contains the raw encoded data from DAQmx. If you have access to LabView you might be able to use this VI to convert DAQmx TDMS files to normal TDMS files: https://decibel.ni.com/content/docs/DOC-32817. I haven't tried that myself though.

PierreSchnizer, 2016-07-04:
Dear All,

    is there still an interest in getting DAQmx working. I started working on reading in the header, which 
seems to work. The NI documentation I saw makes the impression that the data format is sufficiently specified. Did any of you try it recently.

Sincerely yours
  Pierre
  
[tdms.patch.txt](https://github.com/adamreeve/npTDMS/files/346756/tdms.patch.txt)

PierreSchnizer, 2016-07-04:
And If I apply the given patch nptdms seems to read the file. I still have to check if the data is correct ...
[tdms.patch2.txt](https://github.com/adamreeve/npTDMS/files/346787/tdms.patch2.txt)

adamreeve, 2016-09-17:
Hi Pierre

Sorry for the delayed reply. I think a few people would like support for reading DAQmx raw data based on the issues raised about it, and it looks like the NI documentation is a lot more detailed now and supporting it should be possible as you've found. I no longer use LabView myself and don't have any time to spend on implementing and testing this feature. But if somebody else was to implement reading DAQmx raw data, including adding tests for that functionality so we can ensure there aren't regressions in the future, then I'd be happy to merge it.

jshridha, 2016-12-09:
@PierreSchnizer I'm working through your patches and I almost have it working for my raw tdms files. However, I think that I have some issues that I'm not able to sort through. I think the problem is np.fromfile reaches the end of the file when it's trying to read one of the chunks. Is there any chance you can take a look at see if you can help me figure out the issue?

Thanks!

nmgeek, 2016-12-15:
I'm working on using the patched code to read these TDMS files.  I got as far as finding that my sample data files have no samples (the header/metadata reads in without it crashing).  If someone can provide a sample data file with real samples in it I could test this further.

nmgeek, 2016-12-15:
I have a good sample file now.  I reproduced a problem like described by jshridha and I am working on a fix.

nmgeek, 2016-12-16:
While I did find one or two bugs that were not being tripped I think I also found the main problem.  The interleaved data reader is miscalculating the size of a data chunk and reading too much data.  When the code attempts to reshape the oversized array returned from np.fromfile it throws an exception because the array is not an even multiple of the size of one multi-channel sample.

I believe the following update fixes the bug.  I have not yet verified that the new code reads _all_ the data in the TDMS file.  

I also found a couple other problems.  The bigger one is that the spec says there can be more than one 'format changing scaler' but the code in the patch assumes there is always exactly one.  I have no idea what a 'format changing scaler' is, and the data related to them is never used, but the code should handle this so that it doesn't misread data that comes after the scaler specifications.

I will probably play around with the updated code for a while and post either an updated patch or a pull request.
```
         """Read interleaved data where all channels have a numpy type"""
 
         log.debug("Reading interleaved data all at once")
         # Read all data into 1-byte unsigned ints first
         all_channel_bytes = sum((o.data_type.length for o in data_objects))
-        number_bytes = all_channel_bytes * data_objects[0].number_values
+        number_bytes = all_channel_bytes * data_objects[0].tdms_object.number_values
         combined_data = fromfile(f, dtype=np.uint8, count=number_bytes)
```

The above change is in code that is likely used by the legacy TDMS reader so this change could break reading legacy-formatted TDMS files.

nmgeek, 2016-12-16:
The update I proposed above is not correct.  It turns out that the patched code incorrectly takes data_size from the property size of the first 'format changing scaler property'.  The documentation of the TDMS format is not clear but I believe that for raw DAQmx data this data size is fixed at 2 bytes.  (The scaler property data size was 4 bytes, hence the overrun of the file read operation).

My proposed fix is to a) give a byte count of 2 to the tdsTypeDAQmxRawData data type; b) extract that byte count when reading the raw DAQmx metadata section; and c) rename the data members that store the 'format changing scalar' properties (and iterate over them in case there are more than one).

The deltas below are long.  I'll work on a pull request.
```
--- a/nptdms/tdms.py
+++ b/nptdms/tdms.py
@@ -81,7 +81,8 @@ tdsDataTypes.update({
     0x20: DataType('tdsTypeString', None, None, None),
     0x21: DataType('tdsTypeBoolean', 'b', 1, np.bool8),
     0x44: DataType('tdsTypeTimeStamp', 'Qq', 16, None),
-    0xFFFFFFFF: DataType('tdsTypeDAQmxRawData', None, None, None)
+    0xFFFFFFFF: DataType('tdsTypeDAQmxRawData', None, 2, np.uint16)
 })
...
 class _TdmsmxDAQInfo(object):
     __slots__ = [
-        'dimension', 'number_of_values', 'vector_size',
-        'data_type_code', 'data_type',
+        'dimension', 'number_of_values', 'scaler_vector_length',
+        'data_type',
+        'scaler_data_type_code', 'scaler_data_type',
         'raw_buffer_index','raw_buffer_index', 'raw_byte_offset',
         'sample_format_bitmap', 'scale_id',
-        'elements', 'properties'
+        'raw_data_widths', 'properties'
         ]
...
     def _read_metadata(self, f):
         """
+        Read the metadata for a DAQmx raw segment.  This is the raw
+        DAQmx-specific portion of the raw data index.
         """
+        self.data_type = tdsDataTypes[0xFFFFFFFF]
         self.dimension = _read_long(f)
+        # In TDMS file format version 2.0, 1 is the only valid value for dimension
+        assert(self.dimension == 1)
...
-        self.data_type_code = _read_long(f)
-        self.data_type = tdsDataTypes[self.data_type_code]
-        #log.debug(
-        #    "mxDAQ vector vector_size '%d' data_type '%d' = '%s'"
-        #    %(self.vector_size, self.data_type_code, self.data_type.name) )
-        # TBD: this asserts with data type code of 3
-        assert(self.data_type_code == 5)
-
-        self.raw_buffer_index = _read_long(f)    
-        self.raw_byte_offset = _read_long(f)
-        self.sample_format_bitmap = _read_long(f)
-        self.scale_id = _read_long(f)
+        log.debug("mxDAQ vector size '%d'" %(self.scaler_vector_length,))
+
+        for idx in range(self.scaler_vector_length):
+            # tbd: implement format_changing_scaler vector here
+            self.scaler_data_type_code = _read_long(f)
+            self.scaler_data_type = tdsDataTypes[self.scaler_data_type_code]
+            #log.debug(
+            #    "mxDAQ scaler_vector_length '%d' data_type '%d' = '%s'"
+            #    %(self.scaler_vector_length, self.scaler_data_type_code, self.scaler_data_type.name) )
+
+            # more info for format changing scaler
+            self.raw_buffer_index = _read_long(f)    
+            self.raw_byte_offset = _read_long(f)
+            self.sample_format_bitmap = _read_long(f)
+            self.scale_id = _read_long(f)
         #msg = "raw buffer index '%d' raw byte offset '%d' sample_format_bitmap '%d' scale id '%d'"
         #log.debug(msg %(
         #        self.raw_buffer_index,
@@ -881,22 +872,26 @@ class _TdmsmxDAQInfo(object):
         #        self.scale_id
         #))
```

nmgeek, 2016-12-16:
Does anyone want to volunteer to help by writing a regression test or two?

jshridha, 2016-12-16:
Can you share the tdms file you're working with? I'm working through one of my files and it works, but I'm having difficulty getting the raw values to be correct.

jshridha, 2016-12-16:
@nmgeek So it turns out the scale factors for my tdms file are stored in the _TdmsmxDAQInfo object. This object doesn't get saved into memory, so I'm unable to access the polynomial scale factors. I have a couple of commits coming soon that I'll send to your master branch to support these scale factors. I would still appreciate if if you could send that TDMS file. I don't have an example file with linear scale factors.

nmgeek, 2016-12-16:
@jshridha I have not dived into the scaling for my data set.  (We are just starting to evaluate using this raw data format.)  When I looked at the properties attached to each channel in my TDMS file I saw that they were polynomial (linear polynomials, though).

Will my TDMS file help you if the scaling is defined in the properties?

jshridha, 2016-12-16:
@nmgeek Yes, definitely. My data also has the scaling defined in the properties, but I believe our formats may be inherently different. I had to make some minor changes to be able to read the scaling properties for my data and I don't believe you've had to for your data. It would be great if you could share an example tdms file and you can send it to me privately if you want to keep it out of this thread.

nmgeek, 2016-12-16:
Sure.  How do I  contact you privately?

jshridha, 2016-12-16:
@nmgeek Just sent you an email.

nmgeek, 2016-12-19:
jshrida and I worked up the attached pull request which updates npTDMS to read the raw-formatted TDMS files.

jshrida implemented the part that scales the data.  If you use the original .data property you get the raw data.  I wonder if the .data property should give you the scaled data.  Then we could add a .raw or .unscaled property for getting the raw data?

jshridha, 2016-12-19:
I think the implementation should be equivalent with what happens with non-raw TDMS files. If that returns the scaled values, our implementation should also return scaled values.

Based on the example file [here](https://github.com/mnaberez/tdms/blob/master/test/fixtures/example.tdms) I think .data should return the scaled data and have a .raw property with the raw data.

jshridha, 2016-12-19:
I just implemented this if you want to cherry-pick.
https://github.com/jshridha/npTDMS/commit/dade975876b4a90828f132a62b5d7d113a6657c5

jshridha, 2016-12-19:
I'm having a couple of minor issues (likely due to my python version). Line 547 throws an exception since self.num_chunks is a float not an integer. Changing range to np.arange solves that issue.

Second issue is in def fromfile. Count is now also a float and needs to be an int. Casting number_bytes to an int on line 590 fixes that issue.

nmgeek, 2016-12-19:
Excellent.  I might not get to these updates until tomorrow.  Thanks for
finding another sample file.  I also heard there is one with the examples
in LabView.  I have not tracked it down yet.
@adamreeve, we were thinking of checking in some small sample files for
test cases.

adamreeve, 2016-12-19:
Ok cool, using actual files for test cases is fine by me as long as they're not too big.

closed 2017-05-20 by adamreeve