adamreeve/npTDMS #25: Option for getting channel data as dataframe

rubdos, 2015-07-07:
You can get the full tdms file as a Pandas `DataFrame`, but you can't do `as_dataframe` on a channel (which you get by `TdmsFile.object('Group', 'Channel')`).

Is there any interest in making that work? I need this pretty soon. If this isn't implemented yet, I'd give it a go.

adamreeve, 2015-07-08:
Fixed, thanks.

closed 2015-07-08 by adamreeve

rprzysowa, 2017-07-09:
It does not work for me if the channel is not a waveform. 

tdms.object("RMS").as_dataframe()
Traceback (most recent call last):

  File "<ipython-input-36-bdd4f29bce37>", line 1, in <module>
    tdms.object("RMS").as_dataframe()

  File "c:\users\radek\python\nptdms\nptdms\tdms.py", line 756, in as_dataframe
    time = self.time_track(absolute_time)

  File "c:\users\radek\python\nptdms\nptdms\tdms.py", line 665, in time_track
    raise KeyError("Object does not have time properties available.")

KeyError: 'Object does not have time properties available.'

rprzysowa, 2017-07-10:
My version works even if a channel is not a waveform
For a group it adds all its channels to dataframe

def as_dataframe(self, absolute_time=False):
        """
        Converts the TDMS object to a DataFrame

        :param absolute_time: Whether times should be absolute rather than
            relative to the start time.
        :return: The TDMS object data.
        :rtype: pandas.DataFrame
        """

        import pandas as pd

        # When absolute_time is True,
        # use the wf_start_time as offset for the time_track()
        try:
            time = self.time_track(absolute_time) 
        except KeyError:
            time = None
        if self.channel is None:
            return pd.DataFrame.from_items([(ch.channel, pd.Series(ch.data)) \
                        for ch in self.tdms_file.group_channels(self.group)])
        else:
            return pd.DataFrame(self._data, index=time, columns=[self.path])

rubdos, 2017-08-26:
> It does not work for me if the channel is not a waveform.

You might want to open a new issue for this, as this will go unnoticed (as you may have experienced already).

rprzysowa, 2017-08-26:
You're right Ruben but I made a pull request and Adam already handled it,
so the problem is solved
I won't post to closed issues in future
Thanks
Radek

<https://github.com/adamreeve/npTDMS/issues/25#issuecomment-325133611>, or mute
the thread