adamreeve/npTDMS #29: Speeding things up using C/C++?

rubdos, 2015-07-12:
Hi!

Some background: I'm working with a 270MB tdms file for debugging my programs. It takes about 4 seconds for my Python script to do all the work we need on that file. That's already a factor 25 from the previous implementation.

I'm thinking about reimplementing this library (or parts of it, mostly the loading/reading of the file) in C or C++. At first it will be an experiment (am I able to do this? Will it speed things up?).

Would you accept a C/C++ reimplementation, given that the API will stay exactly the same?

adamreeve, 2015-07-14:
I'd be happy to have a C or C++ implementation in npTDMS, but I'd like to keep the pure Python implementation available for when a user doesn't have a C compiler available (eg. on Windows installing C packages is a lot trickier). See cElementTree  vs ElementTree packages for an example of where there's a pure Python and a C version available.

You might also want to consider writing a C library and then a Python wrapper on top of it using [cffi](https://cffi.readthedocs.org/en/latest/) rather than a Python specific library, so that other languages could make use of the C library too?

I do wonder if there's a lot of speed up that could be obtained by refactoring the Python implementation. It does two passes over the file, the first reads all the metdata and creates a whole lot of `_TdmsSegment` objects and then allocates space for all the arrays. Then the second pass actually fills in the data. It's that first pass through that takes all the time. If we could avoid creating all the `_TdmsSegment` objects but still be able to just allocate the data arrays once then reading files should be a lot faster. Or maybe it would be faster to just grow the arrays as data is read rather than doing two passes.

I'm also keen to see how the speed with pypy compares but last time I checked their numpy support was still pretty minimal.

rubdos, 2015-07-15:
I'll try to read and interpret a whole file in a separate C++ library. I'll let you know whether that works and we'll then see how we can integrate them; keeping in mind future integration.

The only thing I can think of which could be annoying that way is the `numpy` integration. There is a C library to communicate with `numpy` and create the objects you have.

DwayneSmurdon, 2015-07-15:
I second the having the ability to wrap it with other langues, I'd really love a Java interface.

rubdos, 2015-07-16:
Would you guys prefer to have a swig interface, or have me code everything by hand?
Any way I would make a separate library in C++ first to see whether it is faster than this implementation.

adamreeve, 2015-07-21:
A SWIG interface would be fine by me, use whatever works best for you.

rubdos, 2015-07-26:
Good news: i finished the reading of the data in a separate C++ module.
Still a lot TODO:
* Big endian is not implemented
* String data is not implemented
* Other tiny checks and bits
* Some data types are not implemented
* Exposure to Python

But the interesting part is:
```
[rsmet@s230ru build]$ time tdmsinfo datafile.tdms
[SNIP]
real    0m1.687s
user    0m1.436s
sys 0m0.253s
[rsmet@s230ru build]$ time tests/tdmsppinfo datafile.tdms
real    0m0.331s
user    0m0.203s
sys 0m0.129s
```

on a 270 MB TDMS file :-)
I think there are still a lot of performance improvements that can be done.

I will now continue making the `tdmsppinfo` executable (which I used as unit test for now) so it prints out the exact same stuff as tdmsinfo does. The `-d` flag already prints about the same as tdmsinfo.

I will continue reporting on this issue report. I'll probably do a `git push` in some minutes to a new repository of mine.

rubdos, 2015-07-26:
I propose using a `git submodule` in this repo to bind TDMSpp to npTDMS, if you agree.

rubdos, 2015-07-27:
Working on the bindings to Python now. Using CFFI and a git-submodule.

It's already prepared so that not using/having CFFI isn't a problem. If CFFI and everything else is found, stuff gets compiled. If not, it's not compiled nor used.

I made a proxy class for TdmsFile and TdmsObject and added an optional constructor argument for TdmsFile to choose the implementation.

[image]

The decreased performance is from loading Python, so when the filesize goes to infinity, performance gain goes to 5.3, if I remember my mathematical limits correctly ;-)

Comments on my C++ code are welcome. I had to use some ugly tricks to make dynamic types work out (`void*`, in particular), but it's not as ugly as I expected. I hope you can live with C++11. I got so used to it that I couldn't do it in an older C++ anymore.

rubdos, 2015-07-28:
You can have a look at my branch here https://github.com/adamreeve/npTDMS/compare/adamreeve:master...rubdos:C_impl?expand=1

Working on the Python bindings now. Still have to think an efficient way to get timestamps from C++ to Python.

adamreeve, 2015-07-28:
Cool that looks like a nice approach to switching out the implementations. Using C++11 seems reasonable and using a submodule also makes sense. :+1:

rubdos, 2015-07-30:
Rearranged your unit tests a little so it runs both implementations.
28 out of 34 tests go well already:

```
ERROR: check_incomplete_data (__main__.cTDMSTestClass)
Test incomplete last segment, eg. if LabView crashed
----------------------------------------------------------------------
Traceback (most recent call last):
  File "nptdms/test/tdms_test.py", line 706, in check_incomplete_data
    tdmsData = test_file.load(implementation=self.implementation)
  File "nptdms/test/tdms_test.py", line 77, in load
    return tdms.TdmsFile(self.file, *args, **kwargs)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/tdms.py", line 50, in __init__
    self.__implementation = cnptdms.TdmsFile(filename)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/cnptdms.py", line 75, in __init__
    handle_exception_if_null(self._ctdmsfile)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/cnptdms.py", line 52, in handle_exception_if_null
    handle_exception()
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/cnptdms.py", line 48, in handle_exception
    raise Exception(ffi.string(what).decode('utf-8'))
Exception: Labview probably crashed, file is corrupt. Not attempting to read.

======================================================================
ERROR: check_interleaved (__main__.cTDMSTestClass)
Test reading interleaved data
----------------------------------------------------------------------
Traceback (most recent call last):
  File "nptdms/test/tdms_test.py", line 442, in check_interleaved
    tdmsData = test_file.load(implementation=self.implementation)
  File "nptdms/test/tdms_test.py", line 77, in load
    return tdms.TdmsFile(self.file, *args, **kwargs)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/tdms.py", line 50, in __init__
    self.__implementation = cnptdms.TdmsFile(filename)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/cnptdms.py", line 75, in __init__
    handle_exception_if_null(self._ctdmsfile)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/cnptdms.py", line 52, in handle_exception_if_null
    handle_exception()
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/cnptdms.py", line 48, in handle_exception
    raise Exception(ffi.string(what).decode('utf-8'))
Exception: Reading inteleaved data not supported yet

======================================================================
ERROR: check_slash_and_space_in_name (__main__.cTDMSTestClass)
Test name like '01/02/03 something'
----------------------------------------------------------------------
Traceback (most recent call last):
  File "nptdms/test/tdms_test.py", line 818, in check_slash_and_space_in_name
    self.assertEqual(len(tdmsData.groups()), 2)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/tdms.py", line 57, in __getattr__
    return getattr(self.__implementation, name)
AttributeError: 'TdmsFile' object has no attribute 'groups'

======================================================================
ERROR: check_string_data (__main__.cTDMSTestClass)
Test reading a file with string data
----------------------------------------------------------------------
Traceback (most recent call last):
  File "nptdms/test/tdms_test.py", line 755, in check_string_data
    tdmsData = test_file.load(implementation=self.implementation)
  File "nptdms/test/tdms_test.py", line 77, in load
    return tdms.TdmsFile(self.file, *args, **kwargs)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/tdms.py", line 50, in __init__
    self.__implementation = cnptdms.TdmsFile(filename)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/cnptdms.py", line 75, in __init__
    handle_exception_if_null(self._ctdmsfile)
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/cnptdms.py", line 52, in handle_exception_if_null
    handle_exception()
  File "/home/rsmet/.local/lib/python3.4/site-packages/npTDMS-0.6.4-py3.4-linux-x86_64.egg/nptdms/cnptdms.py", line 48, in handle_exception
    raise Exception(ffi.string(what).decode('utf-8'))
Exception: Reading string data not yet implemented
[... 26 lines cut ...]
FAIL: check_no_data_section (__main__.cTDMSTestClass)
kTocRawData is set but data length is zero
----------------------------------------------------------------------
Traceback (most recent call last):
  File "nptdms/test/tdms_test.py", line 615, in check_no_data_section
    self.assertTrue(all(data == [1, 2]))
AssertionError: False is not true

----------------------------------------------------------------------
Ran 34 tests in 1.360s

FAILED (failures=2, errors=5)
```

Some of them are easier to fix than others, but I'm getting very close.

Still have some memoryleaks when I'm reading your (pretty tricky) tests too, so I'll have to check on those too.

I also proxied most of the C++ exceptions to Python now, through CFFI.

A little bit about performance:
 * Loading Python modules is slow
 * CFFI seems to be slow too
 * On my 270MB `.tdms` file, I'm having 0m1.713s on Python, 0m1.382s on CFFI, which is pretty terrible. I did _some_ profiling yesterday, but I could probably win some in the C++ part. Most of the time was taken by loading modules (_sigh_)
 * I still think that on bigger `.tdms` files, things will go a lot faster.

adamreeve, 2015-07-30:
Awesome, looks like a good start! Hmm yeah speed up doesn't look great though, can you time the difference in python and check what the times are without module loading?

rubdos, 2015-07-30:
@adamreeve I will time them, just implementing some useful things now. Just pushed `groups()` and `group_channels()` to my branch.

rubdos, 2015-07-30:
Hi @adamreeve, you'll be happy to read this; I did some performance tweaks in by `read_le_float` code (unnecessary copying around stuff, things like that) and I measured:

`INFO:__main__:TDMS: Took 1372.174 ms` for Python
`INFO:__main__:TDMS: Took 590.0530 ms` for CFFI/C++

I'm a happy man again. There probably is a lot more to improve, but read_le_float took 62% of the time
[image]
which I reduced to 8% by now (using `memcpy` instead of copying memory manually). I probably can do that better to.

rubdos, 2015-07-30:
Sorry for spamming your inbox. Just wanted to let you know that array reading has been made a litte faster. Not joking, I have a factor 3.5 in our use case, and if you do not take into account module loading and one-time python stuff like that, there's a factor 5.3.

Now I got it down to 255 milliseconds, I can sleep again. That's more than a gigabyte per second (8 Gbit).

The caveats:
 * Works only on little endian systems when reading LE data (is stated "TODO" about everywhere in my sources), or on BE systems when reading BE data. (All those cases should be taken into consideration, only LE<->LE is implemented atm)
 * Only works on `double` and `float` for now, but can easily be extended to all other types except strings and timestamps and other things that take individual parsing
 * Dataloading to Python/numpy is still somewhat slow IMO, I'd rather have numpy read from my data and not have ownership, but don't know if that's possible with current implementations. I'm doing lazy loading in Python atm.

adamreeve, 2015-07-30:
Awesome, that sounds great. It's probably not necessary to completely implement everything (eg. maybe big endian isn't that widely used in TDMS files?) before getting something merged and providing this as an optional backend. And just supporting ASCII in strings rather than UTF-8 might be reasonable for an initial version.

You should be able to create a numpy array from a pointer to existing data, I've done something similar from a SWIG binding before but not sure how that would work with CFFI. The tricky bit there is just handling data ownership as you say, so that numpy knows not to deallocate the data, but I think that's possible. You might need to look at implementing numpy's array interface: http://docs.scipy.org/doc/numpy/reference/arrays.interface.html

rubdos, 2015-07-31:
> It's probably not necessary to completely implement everything (eg. maybe big endian isn't that widely used in TDMS files?) before getting something merged and providing this as an optional backend.

I'd rather make things more robust first; I still know of one memory leak I have to struggle through. Your unit tests are pretty awesome to find that kind of bugs.

> And just supporting ASCII in strings rather than UTF-8 might be reasonable for an initial version.
I do UTF-8 already, that's no problem. Receiving strings via CFFI generates a bytes object.

Still TODO: test the CFFI stuff in Python 2. More important: test whether Python 2 accepts my class proxying techniques.

That Array interface sounds interesting. I'll have a look at that.

rubdos, 2015-08-01:
Another thing I'm thinking of now, while I'm looking at PEP-3118, the Python buffer interface.

How will I handle this case:
- User creates TdmsFile
- User asks for the numpy data of a channel
- User `del`'s the TdmsFile
- My C++ backend will destroy/`free()` the data of the numpy channel and Python cannot read the array anymore.

Should I use some form of reference counting in my numpy interface? What would you suggest doing, should I add `std::shared_ptr` kind of thing to that memory block, so C++ will delete when it goes out of scope in Python?

rubdos, 2015-08-01:
I just tried using the `__array_interface__`. It seems to speed up just a little (100ms on my 270MB file, which is about 1.5 ns per field (useless information) or 5% (useful unformation)).

It seems to me that pandas is being inefficient with their copying stuff, not me!

I'll check that.

rubdos, 2015-08-01:
@DwayneSmurdon If you're interested in a Java interface for what I have right now, please create a ticket on my https://github.com/rubdos/TDMSpp repository, then we can handle that there. I suggest using swig for that; that will take less time than doing the JNI all by myself and is as performant.

rubdos, 2015-08-01:
Something else I did now: tdmsinfo has a -c/--cffi flag, so that it runs the CFFI backend instead of the Python backend. Optional, of course ;-)

rubdos, 2015-08-03:
Small patch of today: added `wercker.yml` changes for the git submodule.

Somewhere in the future I'll make a pull request to check whether wercker will be able to handle my C++11 stuff... ;-)

DwayneSmurdon, 2015-08-03:
I will do this sometime next week, when I’d have time to try it out.  I’m currently on vacation.  I’m glad to see so much progress being made by you two.

 

-Dwayne

adamreeve, 2015-08-03:
> Somewhere in the future I'll make a pull request to check whether wercker will be able to handle my C++11 stuff... ;-)

The cool thing about wercker is it runs in a Docker container and you can specify any image from the Docker hub to run the tests with, so I've made my own Docker image with Python and numpy installed. I might just need to update it to include gcc. The Dockerfile I'm using is at https://github.com/adamreeve/python-numpy-docker/blob/master/Dockerfile

adamreeve, 2015-08-03:
> * User creates TdmsFile
> * User asks for the numpy data of a channel
> * User del's the TdmsFile
> * My C++ backend will destroy/free() the data of the numpy channel and Python cannot read the array anymore.

Hmm yeah that's tricky. I wonder if it's possible to just transfer ownership to the numpy array somehow so it knows do deallocate the data, without having to do full reference counting? Or get numpy to allocate the data in the first place and then the C++ code writes into the memory location provided by numpy?

As for the documentation, I think the best way to handle this might be to have a single module that has all of the public classes and methods in it, and document that, but those just proxy through to either the pure Python version or the C++ version. Then that also ensures that the public interface is always consistent whether you're using the Python or C++ backend. Is that already what you're doing though?

rubdos, 2015-08-04:
> As for the documentation, I think the best way to handle this might be to have a single module that has all of the public classes and methods in it, and document that, but those just proxy through [...]

Currently, I'm using `getattr` and `__getattr__` stuff, which automagically does what you proposed. But what you said sounds like the sane way of doing things. I'll stick with that.

I'm installing a Windows machine today, not in the mood of programming (been a bit sick too). I'll see if I can fix my memory leak and the documentation this week.

adamreeve, 2015-08-04:
Cool no worries, there's no big rush to get this done, thanks for all your work on it!

HaMF, 2015-09-04:
Thanks a lot for your work guys! (We're also dealing with large TDMS files here and I'm looking forward for your efforts to land (-: )

jshridha, 2016-12-16:
@rubdos Is there any chance you can provide some quick documentation on how to use this implementation?

rubdos, 2016-12-31:
@jshridha It depends on how much information you want; I'd have to reread everything too. The company for which I did this lost interest in this implementation. If you (or anyone else) would have any commercial interest in this, I could finish this.

jshridha, 2016-12-31:
@rubdos I don't need much. Just a quick example of how to compile/use the c-implementation would be great. I understand you've moved on from this and certainly don't need a full-blown documentation.

rubdos, 2016-12-31:
@jshridha try your luck here: https://github.com/rubdos/npTDMS/tree/C_impl (which is my fork, the C_impl branch). I think that if you checkout with submodules, you can just do regular `setup.py`.

adamreeve, 2020-03-17:
I'm going to close this issue as it sounds like work on this has stalled and isn't going to be picked up again, and work on a C/C++ TDMS reader is potentially better done as a separate project.

closed 2020-03-17 by adamreeve

rubdos, 2020-03-17:
Yes, sorry for that.  I'll leave my code online for anyone that ever wants to pick it up!