adamreeve/npTDMS #33: crash on nbsp

FunkyZulu, 2015-07-24:
The library is unable to read a tdms file when there is a non breaking space (0xff) in the property value. I got the following error: 

    UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 196: invalid start byte

This bug does not occur with @rubdos C++ implementation.

rubdos, 2015-07-24:
To give some more context to this issue report:

@FunkyZulu and I are colleagues working on the same job for a month. I tested the bad `.tdms` files using my TDMS implementation in C++, which only reads the metadata for now.

As my implementation doesn't read as utf-8 (yet?), it doesn't complain at it either. Currently it just reads the bytes into an `std::string` and puts them on the heap.

If I print out the properties and their values and use `grep --color='auto' -P -n "\xFF"` to search for that 0xFF value, I can't find it, which bothers me a little.

I'll give priority on fixing this nbsp issue before the C++.

Keeping you posted.

rubdos, 2015-07-24:
This was probably due to file corruption. Sorry for bothering you. I'll reopen the ticket if we start thinking otherwise.

adamreeve, 2015-07-25:
Ok thanks, yeah the documentation for the tdms format says all strings should be utf8, so it does sound like the file was wrong. Maybe we should be able to handle this more gracefully though.

closed 2015-07-25 by FunkyZulu

rubdos, 2015-07-25:
File was corrupt indeed. We can create a new issue for more graceful corruption handling if you're interested.