adamreeve/npTDMS #35: IOError from np.fromfile() when running tests

pdunc, 2015-08-03:
Having an issue running the test suite - this is running in a `venv` with `numpy 1.9.2` and `pandas 0.16.1` installed.

```E

Error
Traceback (most recent call last):
  File "C:\Users\pduncan\AppData\Local\Continuum\Anaconda\lib\unittest\case.py", line 329, in run
    testMethod()
  File "C:\Users\pduncan\PycharmProjects\npTDMS\nptdms\test\tdms_test.py", line 178, in test_data_read
    tdmsData = test_file.load()
  File "C:\Users\pduncan\PycharmProjects\npTDMS\nptdms\test\tdms_test.py", line 77, in load
    return tdms.TdmsFile(self.file, *args, **kwargs)
  File "C:\Users\pduncan\PycharmProjects\npTDMS\nptdms\tdms.py", line 149, in __init__
    self._read_segments(file)
  File "C:\Users\pduncan\PycharmProjects\npTDMS\nptdms\tdms.py", line 183, in _read_segments
    segment.read_raw_data(tdms_file)
  File "C:\Users\pduncan\PycharmProjects\npTDMS\nptdms\tdms.py", line 537, in read_raw_data
    obj._read_values(f, endianness))
  File "C:\Users\pduncan\PycharmProjects\npTDMS\nptdms\tdms.py", line 859, in _read_values
    return np.fromfile(file, dtype=dtype, count=self.number_values)
IOError: first argument must be an open file
-------------------- >> begin captured logging << --------------------
nptdms.tdms: DEBUG: Reading segment at 0
nptdms.tdms: DEBUG: Property kTocDAQmxRawData is False
nptdms.tdms: DEBUG: Property kTocNewObjList is True
nptdms.tdms: DEBUG: Property kTocBigEndian is False
nptdms.tdms: DEBUG: Property kTocInterleavedData is False
nptdms.tdms: DEBUG: Property kTocMetaData is True
nptdms.tdms: DEBUG: Property kTocRawData is True
nptdms.tdms: DEBUG: Reading metadata at 28
nptdms.tdms: DEBUG: Creating a new segment object
nptdms.tdms: DEBUG: Reading metadata for object /'Group'
nptdms.tdms: DEBUG: Object has no data in this segment
nptdms.tdms: DEBUG: Reading 2 properties
nptdms.tdms: DEBUG: Property prop (tdsTypeString): value
nptdms.tdms: DEBUG: Property num (tdsTypeI32): 10
nptdms.tdms: DEBUG: Creating a new segment object
nptdms.tdms: DEBUG: Reading metadata for object /'Group'/'Channel1'
nptdms.tdms: DEBUG: Object data type: tdsTypeI32
nptdms.tdms: DEBUG: Object number of values in segment: 2
nptdms.tdms: DEBUG: Reading 0 properties
nptdms.tdms: DEBUG: Creating a new segment object
nptdms.tdms: DEBUG: Reading metadata for object /'Group'/'Channel2'
nptdms.tdms: DEBUG: Object data type: tdsTypeI32
nptdms.tdms: DEBUG: Object number of values in segment: 2
nptdms.tdms: DEBUG: Reading 2 properties
nptdms.tdms: DEBUG: Property wf_start_offset (tdsTypeDoubleFloat): 0.0
nptdms.tdms: DEBUG: Property wf_increment (tdsTypeDoubleFloat): 0.1
nptdms.tdms: INFO: Read metadata: Took 3.00894216235 ms
nptdms.tdms: INFO: Allocate space: Took 0.060221352437 ms
nptdms.tdms: DEBUG: Reading 16 bytes of data at 241 in 1 chunks
nptdms.tdms: DEBUG: Data is contiguous
nptdms.tdms: INFO: Read data: Took 0.273121545464 ms
--------------------- >> end captured logging << ---------------------
```

rubdos, 2015-08-03:
Looks like the file/stream gets closed while running the read (`np.fromfile`)? I cannot reproduce on current `master` branch.

```
[rsmet@s230ru npTDMS]$ python3 nptdms/test/tdms_test.py
[snip]
Ran 17 tests in 0.071s

OK
```

adamreeve, 2015-08-03:
I can't reproduce this either, what version of Python are you using? And it looks like you're running Windows?

pdunc, 2015-08-04:
Thanks for the responses, after some toying the issue is coming from using the `fromfile` method with a `TemporaryFile`.  I've reported this on the numpy repo at numpy/numpy#6158.

```
OS:     Windows 7 SP1 (x64)
python: 2.7.8
numpy:  1.9.2
```

adamreeve, 2015-08-04:
Hmm yeah looks like it's related to how tempfiles work on Windows. The `tempfile.TemporaryFile` documentation says this:

> The returned object is a true file object on POSIX platforms. On other platforms, it is a file-like object whose file attribute is the underlying true file object. This file-like object can be used in a with statement, just like a normal file.

There's also a `tempfile.NamedTemporaryFile` that behaves more consistently across different platforms, and the returned file-like object has a `file` attribute that is the real open file, so we should be able to use that. I just need to get numpy installed on Windows so I can test this...

commit 0eb31cb0 (2015-08-05): Fix for running tests under Windows

Fixes #35
closed 2015-08-05 by adamreeve