adamreeve/npTDMS #36: Add Support for 'tdsTypeDAQmxRawData'

pdunc, 2015-08-21:
Hi guys,

I tried loading a TDMS file generated using DAQmx file streaming, the data is stored as `tdsTypeDAQmxRawData`.

This throws the `ValueError: Unsupported data type: tdsTypeDAQmxRawData` exception.  Looking at the source there is a section:

```
tdsDataTypes.update({
    ...
    0xFFFFFFFF: DataType('tdsTypeDAQmxRawData', None, None, None)
})
```

Would it be easy to add support for the data?  `pyTDMS` also throws an exception when parsing this file so I wonder if there is something non-trivial about it.

rubdos, 2015-08-21:
Afaik, npTDMS is a numpy re-implementation of pyTDMS, so that would be the non-triviality. I don't have time at this moment (exams), but perhaps @adamreeve can look into this.

adamreeve, 2015-08-22:
Hi, this definitely isn't trivial and I don't have any plans to add support for DAQmx raw data myself. I've left issue #24 open to track this.

The layout of DAQmx raw data was previously undocumented, but NI has added some more information about it at http://www.ni.com/white-paper/5696/en/#toc2. It looks like documentation on the layout of the actual data chunks is still missing though, but someone with experience working with DAQmx data might be able to figure it out based on the metadata provided. It's a lot more complex to read this data than other primitive data types though, it's like a whole other file format embedded within TDMS.

closed 2015-08-22 by adamreeve

pdunc, 2015-08-25:
Hi,

Thanks for the info (and sorry for re-posting a previously answered question)!  I took a look this afternoon at the NI docs and as you mention it is not as simple to achieve.

In my particular circumstance it was much easier to have LabVIEW read the data and dump it to a new TDMS file as a waveform - then npTDMS is happy again, and so am I :ok_hand: