adamreeve/npTDMS #37: Plans for working on XML support

Samuel38, 2015-08-25:
Hi, 
I'm currently working with TDMS files created with Labview and trying to move from DIAdem to Python for data analysis purposes. 
However, it seems the files created both by labview and DIAdem are in my case most definitely XML files. 

Are there any plans to add support for that ?

adamreeve, 2015-08-25:
I'm not personally adding any more features to npTDMS, so I don't plan to add XML support. If somebody wanted to implement that support then I'd be happy to include it though. If you're working on Windows you could try using the Python library based on NI's DLL: https://pypi.python.org/pypi/cTDMS

adamreeve, 2020-04-03:
2020 update: I've been working on tidying up npTDMS recently, fixing some old issues and implementing a few more features. I think it would be worthwhile to support TDM files, but I'm not sure how closely the format maps to the binary TDMS format so this might involve significant work and code changes. There's no documentation on the TDM XML format like there is for TDMS files, but it should be much more straightforward to understand given some example files.

I don't have access to LabVIEW any more so can't easily test this out, and given it's probably a significant amount of work I still don't intend to add this feature, but I'll keep this issue open in case anybody wants to take a look (although nobody has in the last 5 years so it's likely this will just remain open).

aguzmant103, 2020-08-01:
@adamreeve LabVIEW has a Community Edition now! (free) probably still a lot of work though...