adamreeve/npTDMS #39: read only a number of samples?

xaratustrah, 2015-10-08:
Hi!
Would it be possible to read a certain number of samples instead of all?
I did this for a different file format by using  `open`, `seek` and `read` the file only by the amount needed, instead of `np.fromfile`. This sped up operation on large files (>1GB each). Now I am having same file sizes with TDMS and was wondering, if this could be applied here as well? (headers should be read each time). Thanks!

adamreeve, 2015-10-10:
This should be possible but it's not that straightforward due to the segmented structure of TDMS files. You'd probably still need to read through all the metadata of each segment, so the speed up might not be that significant, but memory use should drop. I don't have any plans to implement this myself but I'll leave this issue open in case you or somebody else wants to implement it, because it would be nice to have.

xaratustrah, 2015-10-30:
I finally wrote a hack. Now I can open TDMS files as large as 5GB (more than 4k records per file), jump to the desired position, read desired number of points and return them, without sequentially reading the files.

This is a part of a GUI tool I wrote that that visualizes TDMS files (but also other IQ formats, even WAV files). The code can be found also [here](https://github.com/xaratustrah/iq_suite) on github.

The core idea is to read the first 2 records of the file to estimate sizes that are needed for `f.seek()` along with some other constants. Then in any future run, one can jump as many records as needed.

To determine how many records one needs to read, from a given `starting_sample`, `number_of_samples` I calculate how many full records need to be read. I read those records from the file but return only the desired samples in a simple `numpy complex64` array. This way the whole TDMS thing is transparent to the user.

The whole implementation is just 2 functions withing the file [iqdata.py](https://github.com/xaratustrah/iq_suite/blob/master/iqdata.py), and when removing comments and debug info, it shrinks to a couple of lines. The two functions are called `read_tdms_information` and `read_tdms`, which are used for the first initialization and future reading respectively as described above.

But unfortunately I am not using npTDMS, but just calling the `read_segment` from the original [pyTDMS](http://sourceforge.net/projects/pytdms/files/?source=navbar) of Floris van Vugt **absolutely without** any change.

If you are interested, I would be happy to help, somehow build this simple trick also into npTDMS. There are no dependencies nor other complex structures, just subtle looping and conditional jumps.

cmdcolin, 2016-12-02:
I know there is also a tdms_index format, can that be used to read selected chunks of TDMS files? Thanks for providing the example too @xaratustrah !

fpedaci, 2020-02-16:
Hi (not a specialist here) I'm looking for a way to read just part of large (>10GB) tdms files in python as it can be done in labview, where one can define `counts` and `offset` in [TDMSread](https://zone.ni.com/reference/en-XX/help/371361R-01/glang/tdms_file_read/). 

Is that possible in npTDMS somehow? 
Or @xaratustrah in your `iqdata.py`?

([here](https://stackoverflow.com/questions/60170170/opening-only-part-of-large-tdms-files-in-python-as-labview-does) the stackoverflow question)

adamreeve, 2020-02-16:
Hi, this is still not yet possible with npTDMS but I have started working on it when I can find the time so it should be possible soon.

xaratustrah, 2020-02-17:
Should be possible with [iqtools](https://github.com/xaratustrah/iqtools) library. I regularly use it to jump to a certain position of very large files (tested around 5GB) using `seek` and read a portion only, not the whole file. Generally you should consider the TDMS file containing a huge string of data points with a bit of complex data format inside, **length of frames**, **number of frames** are basically numbers for your analysis convenience. "Jumping" is done using the **starting frame** argument.

```
from iqtools import *
filename ='blah.tdms'
myiq=TDMSData(filename)
myiq.read(nframes=100, lframes=1024, sframes=400)
```
Which means reading 100 frames each 1024 samples long starting from sample 400.

For very small files, there is also another `read_complete_file` function. In the library there are many other tools for spectral analysis such as 1D and 2D spectrograms, **multi taper** etc. There are some examples in the repository.

btw. [iqtools](https://github.com/xaratustrah/iqtools) library has also a GUI front end called [IQGUI](https://github.com/xaratustrah/iqgui), which, although not as versatile as the library interface, can be used to "traverse" through the time file using a visual slider.

maybe this can help.

closed 2020-03-20 by adamreeve

adamreeve, 2020-03-20:
This is now supported in npTDMS as of version 0.22.0. The `read_data` method of `TdmsObject` now takes a length and offset parameter, so you can do something like:
```python
with TdmsFile.open(tdms_file_path) as tdms_file:
    data = tdms_file.object(group_name, chanel_name).read_data(offset, length)
```

fpedaci, 2020-03-20:
this sounds great, thanks a lot.

PR #165 (2020-03-20): Allow reading a subset of channel data
Fixes #39

This adds new `offset` and `length` parameters to the `read_data` method of `TdmsObject` to allow reading only a subset of a channel's data.
commit ad74ff6f (2020-03-20): Allow reading a subset of channel data (#165)

This adds new offset and length parameters to the read_data method of TdmsObject to allow reading only a subset of a channel's data.

Fixes #39