adamreeve/npTDMS #41: Time track incorrect do to failure to read offset

boiler42, 2015-11-19:
I have a tdms file which I know has a time offset for some channels because I can view it in LabView.  However when I get the time with the below it returns an incorrect time.

tdms_obj = tdms_file.object(GROUP, channel)
t = tdms_obj.time_track()

When I looked into it deeper, tdms_obj.property('wf_start_offset') = 0 for all channels, which is incorrect.  Since some of the channels have an offset, when you plot all of the data things don't line up as they should.  I am using relative times throughout.

rubdos, 2015-11-20:
Would you have an example file (as small as possible) that we can look into?

boiler42, 2015-11-20:
I do.  How can I share that with you?

rubdos, 2015-11-21:
I think it's best that @adamreeve looks at this. My knowledge of TDMS files is kinda bad :-)

adamreeve, 2015-11-22:
Your file has offsets of zero for each channel but has different absolute start times. The output from tdmsinfo looks like:
```
/
properties:
name: Test TDMS data
  /'Test Group'
    /'Test Group'/'Channel A'
      data type: tdsTypeDoubleFloat
      properties:
      wf_start_time: 2015-11-20 20:13:43.596313
      wf_start_offset: 0.0
      wf_increment: 1.0
      wf_samples: 10
    /'Test Group'/'Channel B'
      data type: tdsTypeDoubleFloat
      properties:
      wf_start_time: 2015-11-20 20:13:46.596614
      wf_start_offset: 0.0
      wf_increment: 1.0
      wf_samples: 10
```

So when you want to plot the data you could plot each channel against an absolute time or calculate the offset based on the absolute times. If you want to get a time track with absolute times for your channel you can set `absolute_time` to true in the `time_track` method, eg. `t = tdms_obj.time_track(absolute_time=True)`.

boiler42, 2015-11-22:
Understood.  For the real work, I am getting absolute time for each channel and then relativizing them by subtracting the min across all channels.  I think your current code is correct and this is more of a peculiarity of the file I am using.  It would be nice to print a warning if you get relative times from channels with unequal "true start times", which I am considering to be equal to wf_start_time + wf_start_offset.  Thanks for the quick replies and the helpful library!  I am a python guy but where I work all test data is NI/TDMS so nptdms is perfect.

adamreeve, 2015-11-23:
Hmm so re-reading NI's documentation on `wf_start_offset`, they say "This property is used for the LabVIEW Express Dynamic Data Type. Frequency-domain data and histogram results will use this value as the first value on the x-axis."

So this is probably always zero for time-series data. It's just that the `time_track` method can also be used to get an array of x-axis data for non time-series data, so needs to account for this attribute if it's present. This is kind of an unfortunate naming choice, and maybe this method should have just been named "independent_axis_values" or something.

I will update the documentation to clarify what these parameters mean, and note that `wf_start_offset` will be zero for time series data, and the start time should be used to align time-series data.

commit f8efc812 (2015-11-28): Add note about wf_start_offset with time-series data to documentation

Fixes #41
closed 2015-11-28 by adamreeve