adamreeve/npTDMS #42: nptdms crashes python.exe as soon as TdmsFile() is used

jefferyanderson, 2016-01-28:
Hi. I have been using nptdms off and on for a few months and really enjoying it, the team has done great work here.

Unfortunately today I noticed I cannot run it anymore to save my life... any time I try to load a tdms file in it crashes python.exe.

There's very little for me to go on with this, and I've even started from square 1 with a simple:
 from nptdms import TdmsFile
 tdms_file = TdmsFile("D:\\Temp\\In\\test.tdms")

just to see how it works, and it immediately crashes. I've tried with with several files I know worked in the past. What steps can I take to find out the root of this issue? 

I'm running numpy 1.10.4, Windows 7 64-bit, and python3 (64bit)

Thanks and Regards,
Jeff

adamreeve, 2016-01-31:
Hi, what do you mean by "crashes", is there an error message if you run this script from the command line?

jefferyanderson, 2016-01-31:
Yes, I get a Windows message telling me python.exe has crashed.

rubdos, 2016-01-31:
1. Is it this dialog?

[image]

If so, could you paste the data that you can get by clicking `click here`?

2. Are there other Python libraries that crash Python, e.g., did you try doing some plain numpy stuff, without npTDMS? The problem is, a.t.m., there isn't any native code in npTDMS that could possibly make Python give crashes, especially Microsoft dialogs.

jefferyanderson, 2016-01-31:
>    1. Are there other Python libraries that crash Python, e.g., did you

Ruben,
Thanks for the info. I'll try it first thing tomorrow AM.
It's not that particular error, it explicitly complains about python.exe.

Numpy appears to be working correctly, although I know I did recently
uninstall a bunch of NI stuff lately. Does nptdms rely on any NI dlls?
Are there any other places I can check for logs/errors?

Thanks again.
Regards,
Jeff

adamreeve, 2016-02-01:
npTDMS doesn't rely on any NI dlls so that shouldn't be the problem. If you could run your python script from within `cmd.exe` that could give more useful information. There are instructions on how to do this at https://docs.python.org/2/faq/windows.html#how-do-i-run-a-python-program-under-windows.

One thing I noticed about your original example is that you haven't escaped the backslashes in the path you are using. You either need to use double back-slashes, eg. `"D:\\Temp\\In\\test.tdms"` or a raw string by prefixing the string with an r, eg. `r"D:\Temp\In\test.tdms"`.

adamreeve, 2016-04-19:
I'm closing this now, but please reopen if you're still having issues when using an escaped path.

closed 2016-04-19 by adamreeve