adamreeve/npTDMS #51: Refactor Segment metadata reader

nmgeek, 2016-12-17:
@adamreeve and @jshridha, Maybe this is too low level for an issue but I'm not sure where else to log it.
I've been working on the pull request for #24 and feel like the segment metadata reading code is ripe for refactoring.  There are essentially three types of metadata:

- Legacy metadata (not raw format)
- DAQmx raw format metadata (the stuff I am working on for #24)
- Segment metadata with only properties (other data repeats from previous segment)

I envision a property-only metadata class as base class, maybe a base class for the
other two formats, covering the parts they have in common, then two derived classes where one is for Legacy and the other is for Raw format.

This would refactor the `_TdmsSegmentObject._read_metadata` method's big if-then-else
structure (which is effectively a case statement on the `raw_data_index` value).

I'm proposing only refactoring ... no change in functionality.  Ideally we would have more sample TDMS files to test the new DAQmx raw data format.  Because of the limited test data I expect that over time we'll get a couple bug reports and have to apply some updates.  This refactoring will make that process easier for everyone.

What do you think?

adamreeve, 2016-12-17:
Sounds like a good idea to me :+1:

adamreeve, 2020-02-23:
Closing this as I've ended up refactoring this code in #155 and doing something quite similar to what's described here.

closed 2020-02-23 by adamreeve