adamreeve/npTDMS #52: Translator which translates DAQmxRawData files to legacy TDMS files

nmgeek, 2016-12-17:
I think we are close to wrapping up #24 and I see that if you want to read one of these raw-formatted files into Matlab you first have to fire up LabView and translate the raw file into the legacy format.

I think Matlab users would like to have a command line utility which runs outside of LabView so I suggest we include a conversion program that installs in a 'bin' folder like `tdmsinfo`.

How about
````
    tdmsconvert infile.tdms outfile.tdms

adamreeve, 2016-12-17:
This sounds like it would be useful. npTDMS doesn't currently have any support for writing TDMS files though so that would probably need to be implemented first (#11), then this should be quite simple to create.

nmgeek, 2016-12-19:
In the mean time Matlab users might find it useful to have a `tdms2mat` program which saves out a .mat file which can be read by Matlab.  There is a function called `scipy.io.savemat`which can save the .mat file.  I _think_ scipy is included with numpy so it would not introduce a new dependency to nptdms.

I'd like to hear from a Matlab user who would use this.  They would have to have _raw_ formatted TDMS files they wanted to load into Matlab.  (Using LabView to accomplish this seems rather round about.)

I suppose this could be more of a _sample_ program that shows how to use nptdms ...

adamreeve, 2016-12-19:
Scipy builds on top of numpy and isn't included with it, so I'd rather not add it as a dependency. But the code for this is probably simple enough to include as an example in the documentation. I've also just started looking at implementing TDMS writing now so that might not be too far away.

wjlutter, 2017-06-02:
I'm a matlab user that wants to use this package nptdms to read in raw data.  By the way, you can run python in matlab and convert numpy arrays (tdsm data) to matlab array scheme.  Works fine.    I find that I can read in a converted tdms file using imported nptdms  python code  (converted using NI VI object) and the values are OK.  Not so if I read in the unconverted tdms file.   The channel object properties include a linear slope and clearly the nptdms program applies this to the raw data as chanobj.data = chanobj.raw_data*linearslope.    I have an interest in figuring this out (why are the raw data values do not agree with  the  converted values. )   As you just released this 5/17/17, maybe you have an interest as well? 
Bill Lutter (wjlutter@wisc.edu)

adamreeve, 2017-06-07:
Hi @wjlutter. It sounds like converting the TDMS file using the NI VI might strip out the scaling properties so that they're not applied when reading them with npTDMS. Do you not think that applying the scaling on the data property is the correct behaviour for npTDMS?

wjlutter, 2017-06-07:
1.  Well, I used a python script using nptdms  (not invoked from matlab) to read in either the original tdms file or the converted file via the NI VI.
  2.  Reading in the converted tdms file yielded the "proper" values when compared against data in a excel spread sheet using the NI addon that allows excel to read in the original raw tdms files.   So, I knew my python code using the nptdms package was working OK as it agreed with the independent excel output data.   That was very promising.

chan 1 excel spread sheet first 5 values

-0.049580216 -0.032871604 -0.018267334 -0.013101757 0.01827693

chan 1 python program (nptdms) (converted data)

>>>

array([-0.04958022, -0.0328716 , -0.01826733, -0.01310176, 0.01827693,

0.06349629, 0.08059698, 0.07444435, 0.05048364])

Then, I compared the output from the python code applied to the raw tdms file against the excel "proper" values and they were not close.    It was not a multiplicative factor difference, it varied point to point and even the polarity varied in an unsystematic way.

chan 1 python program (nptdms)

>>>

array([ -9.59634781e-05, 1.17897987e-04, -1.78217888e-05,

-1.16527081e-04, 2.74181366e-05, -2.60472298e-05,

-1.22010708e-04, 4.24981117e-05, -5.34653664e-05])

3.   I should add that my supervisor wrote the code producing the tdms files using a 24-bit AD converter.  He says he used standard NI options and did not specifically invoke extended float.   So, he does not think the data from his program is extended float (that nptdms says it cannot handle).   The converted data worked OK.     Do you know a way to test for sure if it is extended float?

4. These are the property (scaled) values from raw tdms for channel 1

a>>> for name, value in tmp2.properties.items():
...     print("{0}: {1}".format(name, value))
...
...
NI_Scaling_Status: unscaled
NI_Number_Of_Scales: 2
NI_Scale[1]_Scale_Type: Linear
NI_Scale[1]_Linear_Slope: 5.3551048040390015e-09
NI_Scale[1]_Linear_Y_Intercept: 0.0
NI_Scale[1]_Linear_Input_Source: 0
NI_ChannelName: Dev2/ai0
unit_string: Volts
NI_UnitDescription: Volts
wf_start_time: 2017-04-02 21:24:35.822570+00:00
wf_increment: 0.0009999999996825864
wf_start_offset: 0.0
wf_samples: 1

Converted file properties for chan 1 are different

wf_start_time: 2017-04-02 21:24:35.822570+00:00

wf_start_offset: 0.0

wf_increment: 0.0009999999996825864

wf_samples: 60658

NI_ChannelName: Dev2/ai0

NI_UnitDescription: Volts

unit_string: Volts

4.   If you were interested, I could record a small file of noise data (to avoid complications for us), verify what I've told you above still holds and send you those files.

Bill Lutter

adamreeve, 2017-06-24:
@wjlutter are you able to share the converted and original TDMS files somewhere? There isn't only a multiplicative factor applied but also an offset so that could possibly account for the polarity change.

wjlutter, 2017-06-26:
I'll put the noise tdms and converted tdms files plus a plot and my python code on box and send you an invite.   What email address do you prefer I send the box invite to?

Bill Lutter

adamreeve, 2017-06-26:
Cool thank you, if you could send that to the email listed here that would be great: https://github.com/adamreeve/npTDMS/blob/master/setup.py#L21

adamreeve, 2020-02-14:
I'm going to close this. If somebody wants to add an example to the documentation showing how to convert data from a TDMS file to Matlab I'd be happy to include that, but I don't think it's worth building into npTDMS itself.

closed 2020-02-14 by adamreeve