adamreeve/npTDMS #60: Accessing attributes

CameronSamuell, 2017-02-02:
First off, this is a great package. It's been a big help to me. 

In writing a TDMS file from labview, it's natural to add in some 'attributes' in addition to the channels and groups. Details of an experiment for example. However, these don't exist like a standard Group/Channel in your framework and so it took me a little while to work out that i could get to them with:

atts = {}
for name, value in TDMS_data.object('Attributes').properties.items():
   atts[name] = value

Perhaps it would be a good idea to have a wrapper around this so a list of attributes can be generated more easily for the newcomer?

adamreeve, 2017-02-05:
Hi, is this "Attributes" object something that's automatically built in to a TDMS file written by LabView or do you create it and name it yourself? I didn't find any documentation on it so I'd prefer to leave things as is.

The properties attribute of a TDMS object is already a dictionary so your code could be simplified to just `attrs = TDMS_data.object('Attributes').properties`, which doesn't seem like too much overhead. I could update the documentation to make it clear that an object doesn't need to be a group or channel but could be an arbitrary object like this with attributes.

CameronSamuell, 2017-02-06:
Hmm. I guess i named it myself (i had in my mind that it was a labview naming). The idea though comes from here: 
http://www.ni.com/white-paper/3727/en/
Where it is broadly a property without a group. Your code addresses it nicely - I think a line in the docs as you describe with your example would be sufficient. 
Thanks!

adamreeve, 2017-02-06:
Ok cool, I've updated the documentation to hopefully make this more clear. One thing that might not have been clear from the npTDMS documentation or that whitepaper is that you can associate properties directly with the file without having to specify a group name. In npTDMS you can get the root file object with `tdms_file.object()`, so the attributes that aren't associated with any group or channel would just be `tdms_file.object().properties`. If you're setting attributes on the root object using the set_properties function in LabVIEW, you just leave out the group and channel options (http://zone.ni.com/reference/en-XX/help/371361J-01/glang/tdms_set_properties/).

closed 2017-02-06 by adamreeve