adamreeve/npTDMS #61: Zero channel data size but non-zero data length based on segment offset

jonititan, 2017-02-09:
I've got some valid data that I can open in labview but can't open using npTDMS.
I get the error message "ValueError: Zero channel data size but non-zero data length based on segment offset"

I've tried doing some trouble shooting but nothing seems to be clarifying the situation yet.
The two files I've been trying 
File Size                   variable "total_data_size"
86,090 KB               88018944
5,604,870 KB          5738590208

adamreeve, 2017-02-19:
Are you able to try commenting out the section of code raising the exception and then see if there's any data missing compared to what LabView reads? The relevant piece of code is at line 383:
```
if total_data_size != data_size:
    raise ValueError(
        "Zero channel data size but non-zero data "
        "length based on segment offset.")
```

It might also be useful to see the debug output by running `tdmsinfo --properties --debug <path to your tdms file>`

lrthomps, 2017-05-09:
I found that by having data_size be calculated from the first non-empty ordered_objects, I don't get the error and all the data is extracted (merely commenting out the error does not yield any data because it thinks there is none when it's just that the first object is an empty pseudo-channel that just needs to be skipped). The relevant change of code is ~line 374 add:
            for i in range(len(self.ordered_objects)):
                data_size = self.ordered_objects[i].number_values \
                    * self.ordered_objects[i].raw_data_width
                if data_size>0:
                    break

closed 2017-05-20 by adamreeve

adamreeve, 2017-05-20:
Thanks for the fix @lrthomps, I've published this change in version 0.11.2.

chaitanya-ather, 2017-07-31:
Hi,

Sorry to dig this thread up again, but I am still facing this issue. 

I have a file with DAQmx data, and when I try to run it,  I get the error mentioned above - I am using v0.11.2.
In tdms.py where the error is raised, total_data_size = 140 while data_size = 0. 

```
>>> from nptdms import TdmsFile
>>> tdms_file = TdmsFile("daqmxdata.tdms")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdms.py", line 94, in __init__
    self._read_segments(f)
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdms.py", line 107, in _read_segments
    f, self.objects, previous_segment)
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdms.py", line 360, in read_metadata
    self.calculate_chunks()
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdms.py", line 393, in calculate_chunks
    "Zero channel data size but non-zero data "
ValueError: Zero channel data size but non-zero data length based on segment offset.
>>>
```
Can you help? I can open a new issue if that is needed.

Thanks,
Chaitanya

adamreeve, 2017-07-31:
Hi @chaitanya-ather yes can you please open a new issue, and also post the output of running `tdmsinfo --debug daqmxdata.tdms`

commit e914786a (2017-05-20): Fix for zero data size in DAQmx raw data

Fixes #61