adamreeve/npTDMS #62: tdmsinfo error

atedalv, 2017-02-27:
Hello, 
npTDMS is a great project and I am using it frequently. Also under linux I am using tdmsinfo to read data file properties and also made my own script to write properties that I need into dat file. However I have found that tdmsinfo with my larger files always reports an Error as following:
`Traceback (most recent call last):
  File "/home/atedalv/.local/bin/tdmsinfo", line 11, in <module>
    sys.exit(main())
  File "/home/atedalv/.local/lib/python2.7/site-packages/nptdms/tdmsinfo.py", line 29, in main
    root = tdmsfile.object()
  File "/home/atedalv/.local/lib/python2.7/site-packages/nptdms/tdms.py", line 155, in object
    raise KeyError("Invalid object path: %s" % object_path)
KeyError: 'Invalid object path: /'`

which for the identical file but with less amount of data (smaller acquisition time) everything works fine. This is the same for npTDMS installed using pip or pip3, sudo or --user. May you please help me with this issue.

adamreeve, 2017-03-04:
Hi, it looks like for some reason these larger files don't have a root object present. I fixed tdmsinfo to handle this and published version 0.10.2 to pypi with the fix.

closed 2017-03-04 by adamreeve

atedalv, 2017-03-06:
Thank you for this fix. The error is gone now; However, for larger tdms files now instead of error I receive as a result of tdmsinfo:  "/"
so no output at all, just  /
shorter tdms files work without any issue.

adamreeve, 2017-03-07:
Ok can you try running tdmsinfo with the "--debug" option and provide the output here?

atedalv, 2017-03-07:
sure, thanks :+1: 
INFO:nptdms.tdms:Read metadata: Took 0.032 ms
INFO:nptdms.tdms:Allocate space: Took 0.00600000000006 ms
INFO:nptdms.tdms:Read data: Took 0.00100000000003 ms
/
I double checked the other long data file and it works with 0.10.2. Seems that only this file doesn't. So I used tdms for excel and it appears that header of this particular file is corrupted. Thus, I assume no problem now. Thank you for the solution of this issue.

adamreeve, 2017-03-18:
Ok thanks for the update. Yeah from the debug output it looks like there must be something wrong with the first header of the file as npTDMS fails to read the first segment and thinks it's reached the end of the file.