adamreeve/npTDMS #64: Scale Data not being applied

racingchuck, 2017-03-31:
Hello,

I have been using npTDMS for while now, and I just realized that if I have only one scaling information in the metadata of a channel it does not get applied to the raw data. I saw in the code that a 1 index is used and Labview starts at zero.

Thank you,

Charles

adamreeve, 2017-04-01:
Hi, can you provide an example of what you mean? An example file with the result you expect would be helpful. Is this for a DAQmx raw data channel? Support for DAQmx raw data was only added recently and there's limited documentation on how it works, so the current implementation probably doesn't support all the features of this type of data.

racingchuck, 2017-04-01:
Hi,

First, thank you for the prompt response. 

When I generate the TDMS file using "TDMS Create Scaling Information" block, the scaling information is added to the channel properties. npTDMS reads the properties correctly but does not scale the data accordingly. Looking in the concerned part of [tdms.py](npTDMS/tdms.py) line 712, property checked for scaled data is 'NI_Scale[**1**]_Scale_Type'. In the file attached (scaled_tdms.tdms) you will see that the property written in the tdms file is 'NI_Scale[**0**]_Scale_Type'. At first I didn't realized the issue because I had the scaling block inside my loop which repeated the scaling information for each segment thus increment the index above 0 (see file repeated_scaling_info.tdms).

The data is not DAQMx. I expect the same result as the repeated_scaling_info.tdms file without having to keep the scaling block in the loop.

**Schema used to create scaled_tdms.tdms**
[image]

**Schema used to create repeated_scaling_info.tdms**
[image]

Thank you,

Charles

***I changed tdms files extension to txt for the upload ***
[repeated_scaling_info.txt](https://github.com/adamreeve/npTDMS/files/887786/repeated_scaling_info.txt)
[scaled_tdms.txt](https://github.com/adamreeve/npTDMS/files/887788/scaled_tdms.txt)

adamreeve, 2017-04-06:
I'm not sure how this should be fixed. We could just check for any scaling property and use the first index found, but that doesn't seem correct. If there are multiple scalings, I assume they should all be applied in order, but I couldn't find any documentation to confirm this. And if I took that approach then your example where you repeatedly apply the scaling with the index incrementing each time would result in a very different result to just having one set of scaling information. I don't have access to LabView any more so can't experiment with how it handles this. Are you able to experiment and see how LabView handles multiple scalings in a TDMS file?

racingchuck, 2017-04-06:
I have access to Labview, I'll run some tests with different scalings configuration and report the results here.

racingchuck, 2017-04-11:
I ran a couple of different tests to isolate how Scaling Information is handled by Labview.
I made a file with two channels with raw data going from 0 to 9.

## **1: No scaling info** 

No scaling info added as a reference
[image]

[image]

## **2: Single scaling info on Channel 1**
Scaling info added on Channel 1 only, scaling infos are written in the channel properties, only Channel 1 is affected.

[image]
[image]

## **3: Single scaling info on Group 1**
Scaling info added on Group 1 only, scaling infos are written in the group properties, all channels in group are affected.

[image]
[image]

## **4: Single scaling info on file**
Scaling info added on file, scaling infos are written in the file properties, all channels are affected.

[image]
[image]

## **5: Multiple scaling info on Channel 1**
Scaling info added on Channel 1 at each iteration, scaling infos are written in the channel properties, only Channel 1 is affected, only last scaling infos are applied.

[image]
[image]

If you want other tests or have any questions let me know.

Files (extension changed from .tdms to .txt for upload):

[no_scaling.txt](https://github.com/adamreeve/npTDMS/files/914022/no_scaling.txt)
[single_scaling_group.txt](https://github.com/adamreeve/npTDMS/files/914005/single_scaling_group.txt)
[multiple_scaling_channel.txt](https://github.com/adamreeve/npTDMS/files/914006/multiple_scaling_channel.txt)
[no_scaling.txt](https://github.com/adamreeve/npTDMS/files/914007/no_scaling.txt)
[single_scaling_channel.txt](https://github.com/adamreeve/npTDMS/files/914008/single_scaling_channel.txt)

adamreeve, 2017-04-19:
Awesome, thanks a lot for that! I won't have time to fix this immediately but will hopefully get around to it soon.

closed 2017-04-23 by adamreeve

adamreeve, 2017-04-23:
This should be fixed in version 0.11.0 (including applying scaling from the group or the whole file)

commit e4a63c06 (2017-04-23): Support different scaling indexes

Fixes #64