adamreeve/npTDMS #66: multiprocessor, multi-thread and nptdms

alphazed35, 2017-04-21:
Hi nptdms, 

I've recently started working with your code.

I am working on .tdms files which are 300GB large - something like 161billion elements in the data. 

With a file this large, doing simple things like finding min and max across that many elements takes some time.. I am wondering about how I would fair accessing the memmap'ed TdmsFile object via multiple threads or on multiple processors.  Is this something you would recommend?

Thanks, 

Nick

adamreeve, 2017-04-23:
Yeah that should be reasonable but there are a couple of things to watch out for. Using multiple threads might not produce any speed up depending on what you're doing, due to the global interpreter lock. If you're mostly calling numpy methods then that should work well as a lot of numpy operations release the GIL. I'd definitely try using multiple threads first and then only try multiple processes if you can't get a speed up using threads. It sounds like there could be some problems with passing the memmapped array into a process using multiprocessing due to issues pickling the array, eg. see http://stackoverflow.com/questions/9964809/numpy-vs-multiprocessing-and-mmap. So you might need to work around that by opening the memmapped file in each of the new processes.

alphazed35, 2017-05-04:
Multiprocessing to load tdms files using npTDMS works well. 

Use the following code as a guide... 

from multiprocessing import Pool
from nptdms import TdmsFile
fileList = [ 
    "probe01.tdms",
    "probe02.tdms",
    .
    .
    .
    "probe24.tdms"
]

def tdmsProcess(fileName):
  print("tdmsProcess for ", fileName," started")
  tdmsObject = TdmsFile(fileName, memmap_dir="./tmp")
  print("tdmsProcess for ", fileName, " complete")
  return tdmsObject, fileName

pool = Pool()
tdmsFiles = pool.map(tdmsProcess, fileList)
pool.close()
pool.join()

tdms object is a list containing the desired npTDMS objects.  Depending on your python you may need to use a pool function .get() of some sort to retrieve value returned from pool.map() - it works for me as is in python3 though.  Watch out... if your tdms files are too large, you will run out of memory at some point. 

nt

closed 2017-05-04 by alphazed35