adamreeve/npTDMS #67: Pickling a memmap'd tdms

alphazed35, 2017-05-04:
Hey Adam, 

Thanks for your response to the previous questions.  

I ended up being able to successfully open 24 tdms files, 12 of which were ~200MB each and 12 of which were ~300GB each; and all that using multiprocessing on 24 Xeon cores... speeding the opening process up quite a bit.  

I assumed that dumping and loading tdmsobjects via pickle would be faster than using TdmsFile to init the tdmsobjects.  The process of pickling worked on the 200MB files, and loading the pickles was indeed faster than initializing a tdmsobject from a tdms file.  

Pickling the 300GB tdmsobjects however causes the pickle module to return a "MemoryError."  I am assuming that this is due to the use of memmaping.  

Any suggestions for how I can get around this and pickle a memmap'd tdmsobject?

nt

alphazed35, 2017-05-04:
I think a related issue, is on pickle.load, how do you bring a serialized object directly into mmap. 

Is this even possible?

alphazed35, 2017-05-04:
So, I believe pickling a memmap'd tdmfile is going to be a problem. 

I believe what I need to do is the following:
-open the 300GB tdms file as a tdmsfile object. 
-isolate the tdms header
-isolate the data and split it into the desired number of chunks x
-write the header and one chunks of data to x new separate tdms files.
-pickle each of the x smaller files. 

The groups/channels info in these files indicates a header with a dict of properties, a single group 'group' and a single channel 'channel1.'

Can anyone quickly cut and paste me some code that would do what I have outlined above for 2 chunks of data?  

Thanks so much,

nt

closed 2017-05-04 by alphazed35

adamreeve, 2017-05-20:
Hi, sorry for the delayed reply. Did you find a way to work around this? I don't think there's a simple way to work around pickling the memmapped file. From your comments in #66 it looks like you went with the approach of opening the TdmsFile within each process instead of passing around the pickled objects.