adamreeve/npTDMS #73: timestamp

rprzysowa, 2017-07-06:
My tdms files consist of an explicit timestamp channel (not implicit sampling information in waveforms). 
As the NI timestamp is incompatible with Python types, npTDMS  has to convert it. But I don't know why the basic type datetime is used instead of numpy datetime64 . Consequently, a timestamp channel in npTDMS is not an array but a list and TdmsWriter is not able to save them.

tdms=TdmsFile('vibration.tdms')
with TdmsWriter(outfile) as tdms_writer:  
    tdms_writer.write_segment(tdms.group_channels('RMS')

The code stops at the object <TdmsObject with path /'RMS'/'Timestamp'> with a list inside it:
{'_data': [datetime.datetime(2017, 6, 4, 14, 4, 28, 966494, tzinfo=<UTC>),
  datetime.datetime(2017, 6, 4, 14, 4, 29, 153948, tzinfo=<UTC>), ...

  286 
    287     try:
--> 288         array.tofile(file)
    289     except (TypeError, IOError, UnsupportedOperation):
    290         # tostring actually returns bytes

AttributeError: 'list' object has no attribute 'tofile'

adamreeve, 2017-07-09:
Thanks for reporting this. Using numpy's datetime64 might have been a better option, I wasn't aware of it at the time, and it would be a backwards incompatible change now. But calling tofile wouldn't work correctly as the way TDMS represents timestamps is different to how numpy does. So the code will need to be changed to handle writing a list of objects.

adamreeve, 2017-07-09:
@rprzysowa I've pushed a fix for this. Can you confirm it works for you? Then I'll create a new release on PyPI.

closed 2017-07-09 by adamreeve

rprzysowa, 2017-07-09:
Yes, my files are written properly. Thank you

commit 1b01a651 (2017-07-09): Fix writing TDMS files with timestamp data

Timestamp channels do not have data stored as a
numpy array so need to be handled differently.

Fixes #73