adamreeve/npTDMS #74: Incorrect data read from DAQmx file

adamreeve, 2017-07-09:
@wjlutter reported this issue in #52. When reading data from a DAQmx data file, the data doesn't match what is obtained if the file is first converted to a normal TDMS file in LabView and then read in. It doesn't appear that this is related to scaling as reading the file without any scaling applied still results in incorrect data, and there isn't an obvious relation between the incorrect and correct data.

[daqmx_tdms_info.txt](https://github.com/adamreeve/npTDMS/files/1133694/daqmx_tdms_info.txt)
[converted_tdms_info.txt](https://github.com/adamreeve/npTDMS/files/1133695/converted_tdms_info.txt)

adamreeve, 2017-07-09:
@wjlutter I haven't been able to figure out what's gone wrong here, but I haven't had much time to look into this. I'm also not that familiar with the DAQmx format and didn't implement DAQmx reading myself. Maybe @nmgeek or @jshridha can see something wrong from the log files above?

wjlutter, 2017-07-10:
Too bad.   Thanks for looking at it.

I notice that in the two attached files, the "string to sign bytes" is the same for the first 70 values then differs.

The "Canonical request bytes" is same for first 274 values then differs.

Bill Lutter

agenecov, 2018-07-20:
@adamreeve @wjlutter I am also getting this issue. However, there are cases in which the code works, and then cases in which the code doesn't. I've tried generating files by logging from two different cDAQ devices; the code works for one of them but only occasionally works the for the other. 

I am setting up the two DAQs both as AI Voltage Channels with linear scalings applied, so I don't think there's anything different in my setup on the software side.

The only thing I can think of that is legitimately different between the two DAQs is the resolution of the ADC: one DAQ has a 16-bit ADC, while the other has a 24-bit ADC. The data gets interleaved, and nptdms reads the data as interleaved numpy.

Here's what I've managed to figure out:

- when logging a single channel from the 16-bit ADC, nptdms parses the file properly
- when logging multiple channels from the 16-bit ADC, nptdms parses the file properly
- when logging a single channel from the 24-bit ADC, nptdms parses the file properly
- when logging multiple channels from the 24-bit ADC, the parsed data is incorrect.

I can provide sample files if you would like.

agenecov, 2018-07-20:
@adamreeve @wjlutter actually I think I just figured out what the problem is. 

in types.py, DaqMxRawData is defined as size=2 and of type np.int16. this works for 16-bit ADCs, as all the data for each sample is contained within 2 bytes. However, for a 24-bit ADC, the size has to increase to 4 / np.int32. So it gets everything approximately right for the first channel (gets the first 16 bits) but completely messes up the second channel (grabs the last 16 bits of the first channel instead of the first 16 bits of the second channel). And so on and so forth for additional channels.

Unfortunately, it doesn't appear that the DaqMxRawData type is specific enough to differentiate between the two data sizes. Additionally, it seems like the minimum raw_data_width, even for a single 16-bit ADC channel, is 4 bytes, which makes doing this programmatically frustrating.

adamreeve, 2018-07-22:
@agenecov thanks for looking into this, it sounds very similar to the problem reported in https://github.com/adamreeve/npTDMS/pull/82/. As you say it doesn't look like there's any way to programmatically determine the data type required for DAQmx data, but maybe adding a way to specify the DAQmx data type when opening the file would be a reasonable workaround.

pradyjowski, 2019-06-21:
Hello,
@adamreeve I found a way to determine the data type. After back-engineering some TDMS files in HEX editor I found the purpose of "raw_data_widths" metadata. It seems npTDMS library just stores this value. 
Firstly it is a vector with unique values for each unique group within a single file. For simplicity let's assume just a single group such that it's a scalar. 
So this value records the total byte length of one record (i.e. byte length of data type * amount of channels). Let me show using two examples:
Example A 
18 channels of PCI-6225 DAQ card logging into single a group
raw_data_widths reads 0x24 = 36 
that means single data type is 36/18 = 2 bytes long => it's either int16 or uint16
Example B
4 channels of NI-9211 logging into a single group
raw_data_widths reads 0x10 = 16
that means single data type is 16/4 = 4 bytes long => it's either int32 or uint32

I know it still doesn't resolve uint vs int, but as you mentioned even NI software asks the user to specify.

Additionally, I confirmed that the only change required to support int32 is in types.py starting line 227:
```
- @tds_data_type(0xFFFFFFFF, np.int16, default_for_nptype=False)
+ @tds_data_type(0xFFFFFFFF, np.int32, default_for_nptype=False)
  class DaqMxRawData(TdmsType):
-    size = 2
+    size = 4
```
I'm lacking the skills in Python to have this check implemented. For now I'm using manually editing the file based on the type of file I'm working with. I'd love to have it implemented though
Let me know if there is any other way I can help.

pradyjowski, 2019-06-25:
Turns out it was easier to implement than I initially thought. 
The code actually correctly reads the raw data as 32bit value, the problem is in how it is subsequently truncated to int16.
I've forked your library and modified it to store daqmxrawdata in int32 values with backward compatibility for int16.  Fork: https://github.com/pradyjowski/npTDMS

adamreeve, 2019-10-14:
Hi @pradyjowski, thanks for looking into this and apologies for neglecting this issue. There is also a "scaler_data_type" property that we read and I'm wondering whether that is the correct data type for the raw data, because it's set to Int16 in the DAQmx data that I have seen. Would you be able to attach your examples here? I think it should then be possible to stop hard coding the data type for DAQmx data and use the DAQmx metadata to determine it, in which case we won't need to convert the data type between int32 and int16.

adamreeve, 2019-10-21:
This should have been fixed with the changes in #132. It looks like the data type ids for DAQmx data don't match the normal TDMS data type ids, which was confusing, but I think now we can work out the DAQmx data type from the metadata.

closed 2019-10-21 by adamreeve