adamreeve/npTDMS #79: Python 3

rddaz2013, 2017-09-14:
Hi

in the python 2 world it works fine ..now with writing .-) great job
But what is with python3?

adamreeve, 2017-09-16:
Python 3 should work, do you have problems using nptdms with Python 3?

rddaz2013, 2017-09-16:
With Python 3 the 'types'-word is map to a other module, i have to rename your 'types' module.  Then the unit-test are working, but I make some additional test with npTDMS in my real workflow. If the "writing" of the TDMS works I am really happy.

It looks ready to use in python3 with this small rename workaround..!

adamreeve, 2017-10-22:
Hi, sorry for the delay in replying to this. I don't have any problem using this with python 3, and as part of the CI build tests are run on Python 3. The types library is also present in Python 2.7. When I import the types module I use `from nptdms import types` so there shouldn't be any conflict with the built in types module. Can you please provide the specific error you are getting and the code or commands you are using that reproduce this error?

closed 2017-10-22 by adamreeve