adamreeve/npTDMS #84: Error writing datetime data that has no time component

adamreeve, 2017-11-19:
Reported in a comment on #18:

> Hi Adam,
> 
> I have problem with datetime object while creating TDMS file. I have tried in so many ways but it could > not work. Can you please help me to solve my issue. Here is my code data object working but data2 object not working

````python
from nptdms import TdmsWriter, RootObject, GroupObject, ChannelObject
import numpy as np
from numpy import datetime64

root_object = RootObject(properties={
"LocoId": 40473,
"Year": 2017,
"Day": 10,
})
group_object = GroupObject("group_1", properties={
"prop1": 1.2345,
"prop2": False,
})

data = np.array([1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0 ,8.0 ,9.0])
dt = np.datetime64('2005-02-25T03:30')
data2 = np.array(['2007-07-13', '2006-01-13', '2010-08-13'], dtype='datetime64')
print(data2)
engine_channel_object = ChannelObject("group_1", "engine_channel_object", data2, properties={
"prop1":"lokesh",
"prop2":"phani",
})

pneumatic_channel_object = ChannelObject("group_1", "pneumatic_channel_object", data, properties={
"prop1":"sundeep",
"prop2":"venkey",
})
LVE_channel_object = ChannelObject("group_1", "LVE_channel_object", data, properties={
"prop1":"sundeep",
"prop2":"venkey",
})
HVE_channel_object = ChannelObject("group_1", "HVE_channel_object", data, properties={
"prop1":"sundeep",
"prop2":"venkey",
})
vehicle_channel_object = ChannelObject("group_1", "vehicle_channel_object", data, properties={
"prop1":"sundeep",
"prop2":"venkey",
})
MISC_channel_object = ChannelObject("group_1", "MISC_channel_object", data, properties={
"prop1":"sundeep",
"prop2":"venkey",
})
datapack_channel_object = ChannelObject("group_1", "datapack_channel_object", data, properties={
"prop1":"sundeep",
"prop2":"venkey",
})
with TdmsWriter("phani_lokesh2.tdms") as tdms_writer:
    tdms_writer.write_segment([
    root_object,
    group_object,
    engine_channel_object, pneumatic_channel_object,LVE_channel_object,HVE_channel_object,vehicle_channel_object,MISC_channel_object,datapack_channel_object])
```

adamreeve, 2017-11-19:
The error looks like:
```
Traceback (most recent call last):
  File "test_datetime.py", line 52, in <module>
    engine_channel_object, pneumatic_channel_object,LVE_channel_object,HVE_channel_object,vehicle_channel_object,MISC_channel_object,datapack_channel_object])
  File "/nptdms/nptdms/writer.py", line 69, in write_segment
    segment.write(self._file)
  File "/nptdms/nptdms/writer.py", line 95, in write
    metadata = self.metadata()
  File "/nptdms/nptdms/writer.py", line 110, in metadata
    metadata.extend(self.raw_data_index(obj))
  File "/nptdms/nptdms/writer.py", line 122, in raw_data_index
    data_type = Int32(obj.data_type.enum_value)
  File "/nptdms/nptdms/writer.py", line 244, in data_type
    return _to_tdms_value(self.data[0]).__class__
  File "/nptdms/nptdms/writer.py", line 278, in _to_tdms_value
    return TimeStamp(value)
  File "/nptdms/nptdms/types.py", line 196, in __init__
    if value.tzinfo is None:
```
it looks like calling `value.astype(datetime)`  behaves in an unexpected way when the value has no time components, and returns a `datetime.date` object instead of `datetime.datetime`.

closed 2017-11-19 by adamreeve

adamreeve, 2017-11-19:
@lokesh422 I've fixed this in the master branch. Can you please test that change and then if it works correctly for you I can push this change to a new version on PyPI.

lokesh422, 2017-11-22:
Thanks a lot Adam.
Now the above code working good. But here one problem is with `datetime`. Actually the `datetime` like  `'2007-07-13 10:15:10'` but in excel file i has changed like `7/13/2007  3:45:00 PM`. I want to get what i'm setting.

Is there any time zone setting in numpy array or anything i need to modify ?

lokesh422, 2017-11-23:
Hello Adam,
It's urgent requirement for me . You are the one who can help me to solve this.

Thanks & Regards,
D Lokesh

lokesh422, 2017-11-23:
Ok Adam, my problem has been solved.
if `datetime `data shoild be like `2017-11-10 23:59:39+0530` it is working good.
But here i need to add `+0530` for every record. If there is any other way without adding `+530` let me know.

adamreeve, 2017-12-01:
@coder422 if you don't specify a timezone for the date values then they will be assumed to be in UTC time, which might be why you were seeing the added 5:30 hours.

And I'm not sure how you're importing your tdms files into Excel, but the formatting of the dates isn't handled by npTDMS, that will be caused by the date format settings in Excel.

adamreeve, 2017-12-02:
The TDMS file has no concept of date formatting, the dates are stored in a binary format. The formatting of your dates will be done by excel. Eg. see https://support.office.com/en-gb/article/Format-a-date-the-way-you-want-8e10019e-d5d8-47a1-ba95-db95123d273e

commit 74ecc86e (2017-11-19): Fix for writing datetime values with no times

Fixes #84