adamreeve/npTDMS #87: Issue with Property Name Capitalization

Ackbach, 2017-12-27:
I've worked up a MWE of something that looks like a bug. I've got a very simple LabVIEW vi, and a very simple Python script. Either file can write a simple numerical array to a tdms file, with a couple of channel-level properties. It appears that when LabVIEW writes a channel, there are three automatic channel-level properties that LabVIEW writes on-the-fly: NI_ChannelLength, NI_DataType, and name. These three data pieces seem to be there whether I open the tdms file in the little native LabVIEW TDMS file viewer, or in Scout, or in DIAdem, or if I import it into Excel using the add-in. However, the npTDMS Python module never returns these three pieces of data in the TdmsFile.object('Group', 'Channel').properties data. I understand these data pieces might be generated on-the-fly, perhaps only when you read the file. Shouldn't the npTDMS module do the same?

[Test npTDMS.zip](https://github.com/adamreeve/npTDMS/files/1589861/Test.npTDMS.zip)

adamreeve, 2018-02-04:
Thanks for the bug report and sorry for the delay replying. Can you also provide a copy of the tdms file that has been generated with LabVIEW with the name property? I no longer have access to LabVIEW so can't run the vi. Not reading a property that's written by LabVIEW definitely sounds like a bug.

I couldn't find any documentation to confirm that the NI_ChannelLength and NI_DataType properties aren't actually written to the file but are generated on the fly when reading, the documentation I could find just says that these properties are readonly, but other readonly properties such as NI_ArrayColumn definitely are written to the file. If this is the behaviour of the NI tools then it probably would be sensible to also include this in npTDMS.

tadekawaw, 2019-03-13:
Hello,
I am having the same issue with not being able to read the NI_ChannelLength property values. Unfortunately, I cannot attach the actual TDMS file due to it containing confidential data, but I have a screenshot of the properties (not the exact file I used in my code, but a similar one), and the output after executing object("Group", "Channel").properties. It looks like it is able to see everything except NI_ChannelLength and NI_DataType. Another thing I noticed is that these properties do not show up in the .index file that is also generated by LabVIEW. However, the Excel Add-in is able to extract the channel length data.
Would you have any suggestions on how I can have my script read these values? Any help would be greatly appreciated.

[image]

[image]

adamreeve, 2019-03-13:
Hi, these properties aren't included in the tdms files explicitly but you can calculate them yourself if needed. `object("Group", "Channel").data_type.enum_value` will give you the type number corresponding to the NI_DataType and `object("Group", "Channel").number_values` will give you the channel length (or `len(object("Group", "Channel").data)`).

tadekawaw, 2019-03-14:
@adamreeve Thank you so much for the quick response! I am now able to get all the values that I need. I truly appreciate your help.

closed 2019-10-12 by adamreeve