adamreeve/npTDMS #91: Number scaling error on import

mgarrett1955, 2018-03-02:
Hello, Thank you for the package!!

Im having and issue reading in a tdms file. I am getting the error:
WARNING:nptdms.scaling:Unsupported scale type: PieceWisePolynomial

I am using Jupyter. I am getting data, that looks like raw data, if this error is encountered, does the return value default to raw?

mgarrett1955, 2018-03-02:
I used the tdmsinfo command line program to read the header information, and can see that the scaling type is piecewise polynomial, which is most likely the extended floating point type not supported by this version. So Im assuming that the raw data got exported. In addition, is there a python function for dumping all of the properties, similar to the command line(I did not take a long time to search, the documentation is great by the way, I just haven't looked!! )Thank again!

adamreeve, 2018-03-04:
Hi, yes in this case the raw unscaled data will be returned instead of the scaled data. There isn't a single function for dumping all properties, but you can use the `properties` attribute of a tdms object to access its properties as a dictionary. It might help to look at how tdmsinfo is implemented: https://github.com/adamreeve/npTDMS/blob/master/nptdms/tdmsinfo.py

closed 2018-07-22 by adamreeve