adamreeve/npTDMS #93: Writing Datetime

ahmetanildindar, 2018-07-17:
Hi,
I'm converting an ASCII file to TDMS format. Here is the ASCII file view as follows
[image]

I'm able to add CH0, CH1 and CH2 columns into TDMS but not "zaman" column with my little python script; 

`from nptdms import TdmsWriter, ChannelObject 
import pandas as pd
#==============================================================================

#%%
file_to_read = "data.txt"

df_veri=pd.read_csv(file_to_read)

#%%

with TdmsWriter("path_to_file.tdms") as tdms_writer:
    data_array1 = df_veri["CH0"].tolist()
    channel1 = ChannelObject('Group', 'Channel1', data_array1)
    data_array2 = df_veri["CH1"].tolist()
    channel2 = ChannelObject('Group', 'Channel2', data_array2)
    data_array3 = df_veri["CH2"].tolist()
    channel3 = ChannelObject('Group', 'Channel3', data_array3)
    data_array4 = df_veri["zaman"].tolist()
    channel4 = ChannelObject('Group', 'Channel4', data_array4 )
    tdms_writer.write_segment([ channel4 , channel1 , channel2 , channel3] )`

The error message I receive is 

> TypeError: unsupported operand type(s) for *: 'int' and 'NoneType'

Here is the screenshot for the code and error message. 

[image]

Any advice is appreciated. 

Best regards from Istanbul. 

++Ahmet

Gygger, 2018-07-19:
The resultant datatype using `.tolist()` on the timestamp column of your DataFrame is `NoneType`, which obviously doesn't fly.  Pandas has a multitude of options to interpret this column as something recognizable to numpy via npTDMS.

Maybe the easiest for you is to try:
`df_veri=pd.read_csv(file_to_read, parse_dates=['zaman'])`

If you want to use non-standard datetime parsing (like converting the whole column to seconds, ms, etc.), you'll have to use `pd.to_datetime` after invoking `pd.read_csv`.

Gygger, 2018-07-19:
Adam didn't comment, I did! :smiley:

I do agree, though, that Adam does deserve much credit for this work.