adamreeve/npTDMS #98: cannot write string to tdms

MayaGoldschmidt, 2018-08-29:
I've been trying to write a string into a tdms file as part of the data, 

```
with TdmsWriter("path_to_file.tdms") as tdms_writer:
    data_array = "trying to add this"
    
    channel = ChannelObject('Group', 'Channel1', data_array)
    tdms_writer.write_segment([channel])
```

but I get the error:

`TypeError: unsupported operand type(s) for *: 'int' and 'NoneType'`

It traces back from the function writer.py:
```
def _data_size(self):
        data_size = 0
        for obj in self.objects:
            if obj.has_data:
                data_size += len(obj.data) * obj.data_type.size
return data_size
```
that in **obj.data_type.size** there is a none type

I tried changing the way I write the string in an array, as a numpy.array and many different ways but none of them seem to help.

What I think the cause of the error is that a size is not defined in types.tdms in the function that defines a string, in every other type there is a size variable except in the string type. 

I'd be happy some assistance

python version 3.6

adamreeve, 2018-08-30:
Hi, unfortunately this was an oversight when implementing tdms file writing, there isn't yet support for writing strings because they work quite differently to other data types and there isn't a constant size for each value. You can add strings as properties but not yet write string data. For example this should work:

```python
with TdmsWriter("path_to_file.tdms") as tdms_writer:
    data_array = [1, 2, 3]
    properties = {"my_string": "trying to add this"}
    
    channel = ChannelObject('Group', 'Channel1', data_array, properties)
    tdms_writer.write_segment([channel])
```

But that might not help you depending on what you're trying to achieve. I'll try to add support for writing string data soon.

adamreeve, 2018-08-30:
I've just pushed a change to allow writing string data. Please try with the latest code from this repository and let me know if you find any problems. Your example won't work as is, you'll need to pass the data as a list:
```
data_array = ["trying to add this"]
```

closed 2018-08-30 by adamreeve

gregstarr, 2018-09-25:
I copy and pasted @MayaGoldschmidt 's code and turned data_array into a list but still get the same error:

TypeError: unsupported operand type(s) for *: 'int' and 'NoneType'

I just downloaded the latest .zip of the repo and installed

commit e349a653 (2018-08-30): Support writing string data

Fixes #98