adamreeve/npTDMS #100: TypeError: Object dtype dtype('O') has no native HDF5 equivalent

Goon83, 2018-09-11:
Hi, 
Met this error related to the type of data. Could anyone have idea to fix it ?

>>> tdms_file.channel_data('Measurement', '0')
array([-218,   93, -183, ..., -105, -204,  -11], dtype=int16)

>>> tdms_file.as_hdf("file.h5", mode='w', group='/')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "//anaconda/lib/python3.5/site-packages/nptdms/tdms.py", line 258, in as_hdf
    container_group.attrs[property_name] = property_value
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper (/Users/dbin/work/soft/h5py/h5py/_objects.c:2686)
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper (/Users/dbin/work/soft/h5py/h5py/_objects.c:2644)
  File "//anaconda/lib/python3.5/site-packages/h5py-2.8.0.post0-py3.5-macosx-10.6-x86_64.egg/h5py/_hl/attrs.py", line 95, in __setitem__
    self.create(name, data=value, dtype=base.guess_dtype(value))
  File "//anaconda/lib/python3.5/site-packages/h5py-2.8.0.post0-py3.5-macosx-10.6-x86_64.egg/h5py/_hl/attrs.py", line 171, in create
    htype = h5t.py_create(original_dtype, logical=True)
  File "h5py/h5t.pyx", line 1614, in h5py.h5t.py_create (/Users/dbin/work/soft/h5py/h5py/h5t.c:19819)
  File "h5py/h5t.pyx", line 1636, in h5py.h5t.py_create (/Users/dbin/work/soft/h5py/h5py/h5t.c:19639)
  File "h5py/h5t.pyx", line 1691, in h5py.h5t.py_create (/Users/dbin/work/soft/h5py/h5py/h5t.c:19540)
TypeError: Object dtype dtype('O') has no native HDF5 equivalent

Bests.
Bin

adamreeve, 2018-09-16:
`dtype('O')` is used for Python objects that are not any of the built in numpy types. Do you have any other channels in the file that are not just int16 data?

russell-merrick, 2019-01-03:
I'm having a similar problem... my data is stored in a single group with multiple channels.  Each channel is a unique datatype.  One is float32, one is int32, one is int16.  Is as_hdf() function supposed to work with different channel data types?

closed 2020-03-17 by adamreeve

adamreeve, 2020-03-17:
Hi, apologies for leaving this for so long. I believe this should be fixed in version 0.21.0, it wasn't the channel data that was the problem but probably that there was a property that was a timestamp (eg. `wf_start_time`), which wasn't handled by  h5py as there's no native support for timestamps in HDF. These are now converted to ASCII strings when writing HDF.

PR #163 (2020-03-17): Fixes for HDF export
Includes changes from #89, but changed to use variable length UTF-8

* Write string data as variable length UTF-8
* Fix writing attributes for groups and channels
* Write timestamps as fixed length ASCII strings, as h5py can't write timestamps directly and HDF has no native support for timestamps. This should fix #100
commit 686a706d (2020-03-17): Fixes for HDF export (#163)

* Write string data as variable length UTF-8
* Fix writing attributes for groups and channels
* Write timestamps as fixed length ASCII strings, as h5py can't write timestamps directly and HDF has no native support for timestamps. This should fix #100