adamreeve/npTDMS #109: Data size is not a multiple of the chunk size

nmgeek, 2019-02-21:
After posting to a closed issue (#40), I am opening this new issue since this is a bug in the latest published release: v0.13.0.  We have tried both an app built against NI's TDMS SDK and the Excel spreadsheet loader and the file loads without errors.  But when you try to load it with nptdms you get the traceback below.

Unfortunately, the TDMS files that exhibit the problem are proprietary.  Does someone know how I might replace all values with zeros to remove the proprietary content?  Then I could post a file.
```
WARNING:nptdms.tdms:Data size 348793088 is not a multiple of the chunk size 2000000. Will attempt to read last chunk
Traceback (most recent call last):
  File "/usr/local/bin/tdmsinfo", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdmsinfo.py", line 26, in main
    tdmsinfo(args.tdms_file, args.properties)
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdmsinfo.py", line 30, in tdmsinfo
    tdmsfile = tdms.TdmsFile(file)
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdms.py", line 94, in __init__
    self._read_segments(f)
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdms.py", line 124, in _read_segments
    segment.read_raw_data(f)
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdms.py", line 512, in read_raw_data
    self._read_interleaved_numpy(f, data_objects)
  File "/usr/local/lib/python2.7/dist-packages/nptdms/tdms.py", line 546, in _read_interleaved_numpy
    combined_data = combined_data.reshape(-1, all_channel_bytes)
ValueError: total size of new array must be unchanged

```

ba05, 2019-02-22:
I don’t know if it would still give you an error but you could open it in labview and multiply all values by random numbers then resave.

adamreeve, 2019-02-23:
You might be able to get more clues about what is going wrong by running `tdmsinfo` on the file with `--debug` and comparing the output with the structure of the data according NI's SDK and the Excel loader.

nmgeek, 2019-02-23:
I played with the source code and was able to read the data.  The file has 16 bytes of junk at the end and the last chunk is incomplete.  It does seem like an ill-formed TDMS file but the producer of the file uses the fact that the NI tools read it without incident as evidence it is a legitimate TDMS file.

I added the code, below, at line 545 of tdms.py.  I'm not sure what terminology to use for these interleaved data bits that are "de-interleaved" in _read_interleaved_numpy.   The logged warning should say that the last of these "bits" was incomplete.  The incomplete "bit" is not a chunk: there are many of these in a chunk.

```
	if len(combined_data) % all_channel_bytes > 0:
            # truncate combined_data so reshape will work                                              
            origlen = len(combined_data)
            nchunk = origlen / all_channel_bytes
            combined_data = combined_data[:nchunk*all_channel_bytes]
            log.warn("Read data not multiple of chunk size. "
                     "Dropping %d bytes from end.", origlen - len(combined_data))
```

joschkazj, 2019-05-13:
Hi,
I was facing a similar issue and created PR #115  (still in draft as I have some files which still fail to load). Just to avoid possible duplicates.

adamreeve, 2020-02-14:
Closing this as it should have been fixed in #115

closed 2020-02-14 by adamreeve