adamreeve/npTDMS #111: digital inputs file:

baudoux, 2019-03-21:
Hello,
I generate a digital input file with nidaqmx as follow.
```
self._task.di_channels.add_di_chan(lines="Dev12/port0/line{}".format(i),
                                                            name_to_assign_to_lines=name,
                                                            line_grouping=LineGrouping.CHAN_PER_LINE)
self._task.start()
...
self._task.stop()
self._task.close()
```
The file opens well in excel but when I try to open it with nptdms.TdmsFile() it raises the following error:
`Unsupported data type to read: <class 'nptdms.types.Void'>`

a simplifide version of the file is on my drive here:
https://drive.google.com/open?id=1owIEUMsXjegMvuo88Nef9_oyznYzeKKG

adamreeve, 2019-10-15:
Hi, apologies for the delay in replying to this. It looks like the problem isn't that npTDMS is trying to read a void data type, but that npTDMS is incorrectly reading the object metadata and so ends up expecting to read a property at an incorrect location in the file. This is caused by your channel having DAQmx data with an undocumented raw data index value. In the [NI documentation](http://www.ni.com/product-documentation/5696/en/), the only raw data index values for DAQmx are `69 12 00 00` and `69 13 00 00`, but your file has `6A 12 00 00`. The DAQmx metadata in your file also appears to have a different, undocumented format, so it isn't clear how to interpret it.

The metadata for the DAQmx channel looks like this (some parts I've guessed the meaning of based on the documented DAQmx format, so these could be wrong):
```
# Length of third object path (40)
28 00 00 00
# Path (/'myGroupName - D13'/'Dev12/port0/line2')
2f 27 6d 79 47 72 6f 75 70 4e 61 6d 65 20 2d 20 44 49 33 27 2f 27 44 65 76 31 32 2f 70 6f 72 74 30 2f 6c 69 6e 65 32 27
# Raw data index (undocumented DAQmx)
6a 12 00 00
# Data type (DAQmx)
ff ff ff ff
# Array dimension (1)
01 00 00 00
# Number of values/chunk size (200)
c8 00 00 00 00 00 00 00
# Scaler vector size (1)
01 00 00 00
# Expect data type (4 bytes), raw buffer index (4 bytes), raw byte offset (4 bytes),
# sample format (4 bytes) and scale id (4 bytes), but only have 17 bytes
00 00 00 00
00 00 00 00
02 00 00 00
00 00 00 00
00
# Raw data width vector size (1)
01 00 00 00
# Raw data width (4)
04 00 00 00
# Number of properties (2)
02 00 00 00
# Property name length (17)
11 00 00 00
# Property name (NI_Scaling_Status)
4e 49 5f 53 63 61 6c 69 6e 67 5f 53 74 61 74 75 73
# Property type (string)
20 00 00 00
# Property length (8)
08 00 00 00
# Property value (unscaled)
75 6e 73 63 61 6c 65 64
# Property name length (19)
13 00 00 00
# Property name (NI_Number_Of_Scales)
4e 49 5f 4e 75 6d 62 65 72 5f 4f 66 5f 53 63 61 6c 65 73
# Property type (U32)
07 00 00 00
# Property value (1)
01 00 00 00
```

adamreeve, 2020-03-28:
I've spent a bit more time looking into this and it looks like the NI documentation is incorrect, `69 13 00 00` is not a valid raw data index value for digital line scaler data, it should be `6A 12 00 00`. It's easy to see how this mistake was made as the digital line scaler value is one more than the format changing scaler value, but in little endian that means the first byte is one more, not the second byte. Testing a file with a raw data index value of `69 13 00 00` in the NI open office TDMS importer doesn't work and logs an error. 

The documentation doesn't explain the difference in format between format changing scaler data and digital line scaler data, but it appears that the "sample format bitmap" is probably only one byte in this case rather than four. (It's unclear what this value represents, it's always zero in the files I've seen and isn't used by npTDMS at the moment).

I've fixed this in #171

PR #171 (2020-03-28): Fix handling of digital line scaler DAQmx data
It looks like the NI documentation (https://www.ni.com/en-in/support/documentation/supplemental/07/tdms-file-format-internal-structure.html) is incorrect and that digital line scaler data uses "6A 12 00 00" rather than "69 13 00 00" in the raw data index. The sample format bitmap also appears to only be one byte instead of four in this case.

Fixes #111
closed 2020-03-28 by adamreeve