adamreeve/npTDMS #117: Reading start measurement time

tom634, 2019-08-26:
How using npTDMS can I read time, at which measurement started ? I can only read time with added time step for each row in tdms_file starting from 0. I want to start it from time, that measurement started i.e. "2019-08-26 15:41" or 15:41:26" or any other time format

tom634, 2019-08-26:
It worked:
pandas_dataframe=tdms_file.as_dataframe(time_index=True, absolute_time=True) # Convert the TDMS object to Pandas array

but what should I do if I would like to have Time in separate column, not index column?

tom634, 2019-08-26:
It worked:
pandas_dataframe=tdms_file.as_dataframe(time_index=True, absolute_time=True) # Convert the TDMS object to Pandas array

Then if I want to have time as separate column and reindexed index, I should type:

df['Time'] = df.index
df.index = pd.RangeIndex(len(df.index))

But event with this option I can't use matplotlib fully, so the line:
ax[1][0].tripcolor(b_1_experiment_probe1,b_2_experiment_probe1,z_A_s_probe1,shading='gouraud',cmap='jet')
doesn't work. If I comment this line, all other things work. This line doesn't work only if time_index=True.

Traceback (most recent call last):
  File "c:\Users\Tomek\.vscode\extensions\ms-python.python-2019.8.30787\pythonFiles\ptvsd_launcher.py", line 43, in <module>
    main(ptvsdArgs)
  File "c:\Users\Tomek\.vscode\extensions\ms-python.python-2019.8.30787\pythonFiles\lib\python\ptvsd\__main__.py", line 432, in main
    run()
  File "c:\Users\Tomek\.vscode\extensions\ms-python.python-2019.8.30787\pythonFiles\lib\python\ptvsd\__main__.py", line 316, in run_file
    runpy.run_path(target, run_name='__main__')
  File "C:\Users\Tomek\Anaconda3\lib\runpy.py", line 263, in run_path
    pkg_name=pkg_name, script_name=fname)
  File "C:\Users\Tomek\Anaconda3\lib\runpy.py", line 96, in _run_module_code
    mod_name, mod_spec, pkg_name, script_name)
  File "C:\Users\Tomek\Anaconda3\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "c:\Users\Tomek\Documents\Experiment\process_data_interpolation.py", line 736, in <module>
    ax[1][0].tripcolor(b_1_experiment_probe1,b_2_experiment_probe1,z_A_s_probe1,shading='gouraud',cmap='jet')
  File "C:\Users\Tomek\Anaconda3\lib\site-packages\matplotlib\axes\_axes.py", line 8107, in tripcolor
    return mtri.tripcolor(self, *args, **kwargs)
  File "C:\Users\Tomek\Anaconda3\lib\site-packages\matplotlib\tri\tripcolor.py", line 53, in tripcolor
    tri, args, kwargs = Triangulation.get_from_args_and_kwargs(*args, **kwargs)
  File "C:\Users\Tomek\Anaconda3\lib\site-packages\matplotlib\tri\triangulation.py", line 167, in get_from_args_and_kwargs
    triangulation = Triangulation(x, y, triangles, mask)
  File "C:\Users\Tomek\Anaconda3\lib\site-packages\matplotlib\tri\triangulation.py", line 55, in __init__
    self.triangles, self._neighbors = _qhull.delaunay(x, y)
RuntimeError: Error in qhull Delaunay triangulation calculation: input inconsistency (exitcode=1); use python verbose option (-v) to see original qhull error.

tom634, 2019-08-26:
How can I export to full dataframe absolute time only for single, selected channel and omit times for other channels? For other channels the time should not be readed and treated as NaNs. Is it possible?

adamreeve, 2020-03-21:
Hi, apologies for not replying earlier. Usually when using a dataframe all values in a row are related, in the case of time series data this means they're usually measurements for the same point in time. So if in your case this isn't true when not using a time index due to different measurement frequencies, it sounds like it might be better to convert channels to dataframes separately.

Eg. you can use `df = tdms_file.object(group_name, channel_name).as_dataframe(absolute_time=True)` for the channel you want a time index for.

Note you can also get the time index for a channel as an array without converting to pandas with `tdms_file.object(group_name, channel_name).time_track(absolute_time=True)`. Or if all you really want is to just find the start time for a channel, use `tdms_file.object(group_name, channel_name).properties['wf_start_time']`.

adamreeve, 2020-03-28:
I'm going to close this as I think it's been answered now but feel free to reopen if you have further questions.

closed 2020-03-28 by adamreeve