adamreeve/npTDMS #118: Use a child logger for logging

ojgc, 2019-09-04:
Hello, I think the use of the logging module in tdms.py is slightly broken, as it relies on reconfiguring the root logger which doesn't work well when using logging in other modules that include tdms.  A better solution would be for tdms to create a child logger and use that.

```
import logging
log = logging.getLogger(__name__)
logging.basicConfig()
log.setLevel(logging.WARNING)
```
The above is the relevant code from tdms.py.  Other modules using logging that import tdms are unable to reconfigure logging, since subsequent calls to logging.basicConfig() have no effect.  eg. if I then add the following lines copied straight from https://docs.python.org/3/howto/logging.html to the above

```
logging.basicConfig(format='%(asctime)s %(message)s')
logging.warning('is when this event was logged.')
```
the output is:
```WARNING:root:is when this event was logged.```
rather than the expected:
```2010-12-12 11:41:42,612 is when this event was logged.```

I'd like to propose a code change to use a child logger for tdms.py, based on the example code in the howto, this would replace lines 26-28 in tdms.py:
```
#log level
log_level = logging.WARNING

# create logger
log = logging.getLogger('tdms')
log.setLevel(log_level)

# create console handler and set level to debug
ch = logging.StreamHandler()
ch.setLevel(log_level)

# create formatter
formatter = logging.Formatter('%(name)s - %(levelname)s - %(message)s')

# add formatter to ch
ch.setFormatter(formatter)

# add ch to logger
log.addHandler(ch)
```
Calls to log.error(), etc. later on in the code can remain as is.  This will allow other modules to use logging too without interfering with each other.

adamreeve, 2019-09-06:
Hi, thanks for pointing out this issue, I'd be happy to accept a pull request with this change.

ojgc, 2019-09-10:
OK, thanks!

closed 2019-09-28 by adamreeve