adamreeve/npTDMS #120: Incorrect data read from the tdms file

gkm147, 2019-09-19:
The data I read from the a tdms file using npTDMS is completely different from the data when I read using the pd.read_excel and it seems that the data from the latter option is correct.

adamreeve, 2019-09-28:
Hi @gkm147, it's a bit hard to help you unless you can provide an example file and what you expect the data to be

gkm147, 2019-09-30:
Hi Adam,

I attached an example TDMS files saved as an Excel file below. The data is column B is approximately -48611 in the 'Data (Scaled)' Tab. 
[Example.xlsx](https://github.com/adamreeve/npTDMS/files/3672573/Example.xlsx)

However, when I read the TDMS file using the npTDMS, it reads the following data(see attached image) which is not the same as the data in the Excel file. 

[image]

adamreeve, 2019-10-02:
It looks like the data is off by a constant scaling factor, so  there's probably a bug in how the scaling is being applied.

Is it possible to attach the tdms file itself, or can you post the output of running `tdmsinfo --properties' on the tdms file? The tdmsinfo command should be installed when you install the nptdms package.

I would guess the timestamp difference is due to timezones, the time that npTDMS returns will be in UTC but possibly your excel file is showing local times?

gkm147, 2019-10-03:
The timestamp makes sense now. Attached below is the tdms files itself.  

[Example.zip](https://github.com/adamreeve/npTDMS/files/3686232/Example.zip)

adamreeve, 2019-10-03:
Thanks, I can see what the problem is now. The output from tdmsinfo shows this for one of the channels for example:
```
    /'Data (Scaled)'/'RSG1'
      data type: DoubleFloat
      properties:
      NI_UnitDescription: microstrain
      unit_string: microstrain
      Signal_Type: RSG
      NI_Scaling_Status: unscaled
      NI_Number_Of_Scales: 2
      NI_Scale[0]_Scale_Type: Linear
      NI_Scale[0]_Linear_Slope: 2.5
      NI_Scale[0]_Linear_Y_Intercept: 0.0
      NI_Scale[0]_Linear_Input_Source: 4294967295
      NI_Scale[1]_Scale_Type: Polynomial
      NI_Scale[1]_Polynomial_Coefficients_Size: 2
      NI_Scale[1]_Polynomial_Coefficients[0]: 0.0
      NI_Scale[1]_Polynomial_Coefficients[1]: 778.08
      NI_Scale[1]_Polynomial_Input_Source: 0
      NI_ArrayColumn: 0
```
The data has two scalings, one linear and one polynomial, and npTDMS is only applying the second scaling but it should be applying them both in order, so this channel data is out by a factor of 2.5.

adamreeve, 2019-10-03:
This should be fixed now, I've released version 0.16.0 with a fix to apply all the scalings

PR #122 (2019-10-03): Support multiple scalings
Fixes #120
closed 2019-10-03 by adamreeve