adamreeve/npTDMS #125: DaqMXRawData format support

jmd1979, 2019-10-08:
Hi,
I files that were created from a data acquisition system, with a raw type. Is there plans to support this data type? Here is the output from tdmsinfo.  

`[nptdms.tdms    ERROR] mxDAQ multiple format changing scalers not implemented
properties:
name: TC file
  /'Data'
    /'Data'/'TC1'
      data type: DaqMxRawData
      properties:
      NI_Scaling_Status: unscaled
      NI_Number_Of_Scales: 13
      NI_Scale[1]_Scale_Type: Linear
      NI_Scale[1]_Linear_Slope: 9.31322630127e-09
      NI_Scale[1]_Linear_Y_Intercept: 0.0
      NI_Scale[1]_Linear_Input_Source: 0
      NI_Scale[3]_Scale_Type: Linear
      NI_Scale[3]_Linear_Slope: 9.31322630127e-09
      NI_Scale[3]_Linear_Y_Intercept: 0.0
      NI_Scale[3]_Linear_Input_Source: 2
      NI_Scale[4]_Scale_Type: Subtract
      NI_Scale[4]_Subtract_Left_Operand_Input_Source: 1
      NI_Scale[4]_Subtract_Right_Operand_Input_Source: 3
      NI_Scale[5]_Scale_Type: Linear
      NI_Scale[5]_Linear_Slope: 1000000.0
      NI_Scale[5]_Linear_Y_Intercept: -0.0
      NI_Scale[5]_Linear_Input_Source: 4
      NI_Scale[7]_Scale_Type: Linear
      NI_Scale[7]_Linear_Slope: 2.98023241641e-07
      NI_Scale[7]_Linear_Y_Intercept: 0.0
      NI_Scale[7]_Linear_Input_Source: 6
      NI_Scale[8]_Scale_Type: Table
      NI_Scale[8]_Table_Scaled_Values_Size: 206
      NI_Scale[8]_Table_Scaled_Values[0]: 2.449179
      NI_Scale[8]_Table_Scaled_Values[1]: 2.445375
      NI_Scale[8]_Table_Scaled_Values[2]: 2.441324
      NI_Scale[8]_Table_Scaled_Values[3]: 2.437012
      NI_Scale[8]_Table_Scaled_Values[4]: 2.432426
      NI_Scale[8]_Table_Scaled_Values[5]: 2.427554
      NI_Scale[8]_Table_Scaled_Values[6]: 2.422379
      NI_Scale[8]_Table_Scaled_Values[7]: 2.416889
      NI_Scale[8]_Table_Scaled_Values[8]: 2.411068
      NI_Scale[8]_Table_Scaled_Values[9]: 2.4049
      NI_Scale[8]_Table_Scaled_Values[10]: 2.398371
      NI_Scale[8]_Table_Scaled_Values[11]: 2.391463
      NI_Scale[8]_Table_Scaled_Values[12]: 2.384163
      NI_Scale[8]_Table_Scaled_Values[13]: 2.376453
      NI_Scale[8]_Table_Scaled_Values[14]: 2.368317
      NI_Scale[8]_Table_Scaled_Values[15]: 2.359739
      NI_Scale[8]_Table_Scaled_Values[16]: 2.350703
      NI_Scale[8]_Table_Scaled_Values[17]: 2.341193
      NI_Scale[8]_Table_Scaled_Values[18]: 2.331192
      NI_Scale[8]_Table_Scaled_Values[19]: 2.320684
      NI_Scale[8]_Table_Scaled_Values[20]: 2.309655
      NI_Scale[8]_Table_Scaled_Values[21]: 2.298087
      NI_Scale[8]_Table_Scaled_Values[22]: 2.285968
      NI_Scale[8]_Table_Scaled_Values[23]: 2.273283
      NI_Scale[8]_Table_Scaled_Values[24]: 2.260018
      NI_Scale[8]_Table_Scaled_Values[25]: 2.246159
      NI_Scale[8]_Table_Scaled_Values[26]: 2.231696
      NI_Scale[8]_Table_Scaled_Values[27]: 2.216619
      NI_Scale[8]_Table_Scaled_Values[28]: 2.200916
      NI_Scale[8]_Table_Scaled_Values[29]: 2.184578
      NI_Scale[8]_Table_Scaled_Values[30]: 2.1676
      NI_Scale[8]_Table_Scaled_Values[31]: 2.149975
      NI_Scale[8]_Table_Scaled_Values[32]: 2.1317
      NI_Scale[8]_Table_Scaled_Values[33]: 2.112769
      NI_Scale[8]_Table_Scaled_Values[34]: 2.093185
      NI_Scale[8]_Table_Scaled_Values[35]: 2.072945
      NI_Scale[8]_Table_Scaled_Values[36]: 2.052056
[... 376 lines cut ...]
      NI_Scale[8]_Table_Input_Source: 7
      NI_Scale[9]_Scale_Type: Linear
      NI_Scale[9]_Linear_Slope: 1.0
      NI_Scale[9]_Linear_Y_Intercept: 0.0
      NI_Scale[9]_Linear_Input_Source: 8
      NI_Scale[10]_Scale_Type: Thermocouple
      NI_Scale[10]_Thermocouple_Thermocouple_Type: 10073
      NI_Scale[10]_Thermocouple_Scaling_Direction: 1
      NI_Scale[10]_Thermocouple_Input_Source: 9
      NI_Scale[11]_Scale_Type: Add
      NI_Scale[11]_Add_Left_Operand_Input_Source: 5
      NI_Scale[11]_Add_Right_Operand_Input_Source: 10
      NI_Scale[12]_Scale_Type: Thermocouple
      NI_Scale[12]_Thermocouple_Thermocouple_Type: 10073
      NI_Scale[12]_Thermocouple_Scaling_Direction: 0
      NI_Scale[12]_Thermocouple_Input_Source: 11
      NI_ChannelName: TC1
      unit_string: Deg C
      NI_UnitDescription: Deg C
      wf_start_time: 2019-04-25 22:45:37.953401+00:00
      wf_increment: 1.0
      wf_start_offset: 0.0
      wf_samples: 1

`

adamreeve, 2019-10-12:
Hi, I don't have a lot of time to work on npTDMS any more so it's unlikely this will be added any time soon unless somebody else has a go at implementing it. I'd be happy to accept a pull request for this. When I originally looked into this there wasn't much documentation on the DAQmx format but I think that has improved now.

adamreeve, 2019-10-13:
It would help if you could provide an example file that has that error, along with the data you would expect to be read.

jmd1979, 2019-10-18:
Here is a zipped file that contains analog input, TC, and RTD data. The tdms and a converted csv file are included. 

I've looked into working on this issue, but I do more data analysis work. I sorta understand what the tdms code is doing, but I'm not proficient enough make changes. If you could offer a suggestion on what functions need modification, I will try to get it working.

Regards,
John    

[DaqMXRawData_example.zip](https://github.com/adamreeve/npTDMS/files/3743972/DaqMXRawData_example.zip)

adamreeve, 2019-10-19:
Thanks, for the RTD file it should be straightforward to add a new scaling type in `nptdms/scalings.py`, see the `_read_linear_scaling` and `_read_polynomial_scaling` functions for example which use the scaling properties and return scaling objects that know how to apply the scaling. The RTD scaling looks a bit more complicated. There is documentation on the different scaling types at http://zone.ni.com/reference/en-XX/help/371361R-01/glang/tdms_create_scalinginfo/. For the RTD scaling it looks like you need to use the Callendar-Van Dusen equation to convert a voltage into a temperature.

I'll take a look at the TC file and see if I can make sense of the scaler information. The TDMS documentation now describes the layout of this data but doesn't give any details of how to interpret it. At the moment the DAQmx reading in npTDMS is very limited and this scaler information is ignored.

adamreeve, 2019-10-20:
Looking more into the scaling, this looks quite complicated and would need more work than just implementing the different scaling types. It looks like some of the scalings refer to outputs from other scalings and we'd need to interpret a graph of scaling expressions. At the moment the scale inputs aren't used and we just apply scales in order.

The full scaling graph looks something like:
[image]

What's confusing me is how to deal with the undefined inputs. scale 1 and 3 have the same slopes so if scale 2 being undefined means it refers to the raw data, then we'd be subtracting the data from itself and then end up with zeros. So maybe if scale 2 is undefined we fall back to the output from scale 1? Do you know how these scalings should be applied? Once the raw data is read correctly then hopefully it's straightforward to confirm how the scaling should work.

adamreeve, 2019-10-20:
Ah the undefined scales are actually referring to the scalers from the DAQmx metadata. The DAQmx metadata looks like this:

```
DaqMxMetadata(
        data_type=<class 'nptdms.types.DaqMxRawData'>,
        dimension=1,
        chunk_size=0,
        raw_data_widths=[12],
        scalers=[
        DaqMxScaler(scale_id=0, scaler_data_type=<class 'nptdms.types.Uint8'>, scaler_raw_buffer_index=0, scaler_raw_byte_offset=0, scaler_sample_format_bitmap=0),
        DaqMxScaler(scale_id=2, scaler_data_type=<class 'nptdms.types.Uint8'>, scaler_raw_buffer_index=0, scaler_raw_byte_offset=4, scaler_sample_format_bitmap=0),
        DaqMxScaler(scale_id=6, scaler_data_type=<class 'nptdms.types.Uint8'>, scaler_raw_buffer_index=0, scaler_raw_byte_offset=8, scaler_sample_format_bitmap=0)])
```

So the DAQmx data stores 12 bytes for each data point. It looks like byte 0 is scale 0, byte 4 is scale 4 and byte 8 is scale 6, and the other bytes are all unused as the data types are all Uint8.

Getting this to work is going to be fairly complicated and require a bit of refactoring of the code I think.

adamreeve, 2019-10-21:
I think with my latest changes these files should work now once the unimplemented scaling types are added. The multiple scalers should now work, and I realised there was a bug in how data types for the scaler are calculated, so the data isn't actually Uint8 but is Int32, which makes more sense as there aren't any unused bytes.

adamreeve, 2019-11-19:
I'm going to close this as fixed as with the latest npTDMS release (0.18.1) both these files are mostly read correctly.  The only difference is that with the thermocouple scaling file, there are some values that are outside the valid range for a K type thermocouple that result in a NaN temperature, whereas the converted CSV does have a non-NaN value for these.

I'm using the [thermocouples_reference package](https://pypi.org/project/thermocouples_reference/) to implement the thermocouple scaling and this throws an exception for values out of range, which I'm catching and then returning a NaN value. I think this is quite reasonable rather than returning an extrapolated and inaccurate value.

I also had an issue installing this package in Python 3, getting a UnicodeDecodeError, but I could work around this by first setting the locale to UTF-8:
```
LC_ALL=C.UTF-8 pip install thermocouples_reference
```

I've also only partly implemented RTD scaling. Supporting negative temperatures would be much more complicated and will raise an exception for now. I'm also not accounting for lead wire resistance as that is untested and an exception is raised if this is non-zero.

closed 2019-11-19 by adamreeve