adamreeve/npTDMS #142: wrong values of timeStamps

HenningScheufler, 2019-12-06:
Hi,

When i load a tdms file the timeStamp information is incorrect compared to Diadem 2017. The data was recorded at the 6.12.2019 at 13.16 but i get the value:
 2019-12-06T12:16:27.550994 
which is on hour earlier suggesting that it is related to the timezone (Bremen, Germany GMT +1)

I attached the example file:
[timeStamp.zip](https://github.com/adamreeve/npTDMS/files/3932185/timeStamp.zip)

Best

Henning

adamreeve, 2019-12-06:
Hi, yes TDMS stores timestamps in UTC which is what npTDMS returns. Diadem must be displaying times in your local timezone when viewing the file. This will be the timezone of the machine where you are viewing the data, not the local timezone where the data was generated, which isn't stored in the file. So if you were to open the same file in a different timezone the time would be different again.

If you want to view timestamps in your local timezone in Python you will need to convert them from UTC. You can use the arrow library for this, for example:
```python
import datetime
import arrow
import numpy as np

timestamp = np.datetime64('2019-12-06 20:40')
local_time = arrow.get(timestamp.astype(datetime.datetime)).to('local')
print(local_time.format())
```

This is a little bit awkward as you have to first convert the numpy datetime64 to a Python datetime using `astype` before using arrow to convert to local time.

I will add something to the documentation to make this clear.

HenningScheufler, 2019-12-10:
Would it be possible to to add a function to nptdms that updates the timestamp to the local time?

adamreeve, 2019-12-10:
No I think that's outside of the scope of what npTDMS should be responsible for.

closed 2020-02-08 by adamreeve