adamreeve/npTDMS #143: Support for `with` keyword

gsainsbury86, 2019-12-19:
It's possible that I've missed something, but is it possible to support `with` when reading from a TDMS file?

I have a flask front end which is queried often and is falling over after a few hours. I think it might be because file handles aren't being cleared up properly.

I can't see a close method for TdmsFile, and this snippet produces the following result

```
with nptdms.TdmsFile("example.tdms") as tdms_file:
    print(tdms_file)
```

```
Traceback (most recent call last):
  File "wsn_read_test.py", line 4, in <module>
    with nptdms.TdmsFile("/Volumes/wsn-data-repository/2019-12-18.tdms") as tdms_file:
AttributeError: __enter__
```

gsainsbury86, 2019-12-19:
Reading a bit more closely, it looks like the file handle is being closed upon reading the entire file into memory when the object is created, so maybe this isn't necessary and my root problem is somewhere else.

adamreeve, 2019-12-21:
Hi, yes when constructing a TdmsFile everything is read up front and the file is closed after reading (if a path is passed to the constructor rather than an open file), so there's no need for TdmsFile to be a context manager.

closed 2019-12-21 by adamreeve