adamreeve/npTDMS #148: tdms with NI-XNET CAN frames

ThetomekK, 2020-02-14:
Hi there,

i have several hundrets of .tdms files logged with a cRIO and a CAN Interface. 
nptdms can read the files, but all data is interpreted as unint8, so without any decoding of the CAN frames, as expected.

Are there any plans to extend ntdms to be able to read this type of .tdms files?
There is a lot of documentation:

[http://zone.ni.com/reference/en-XX/help/372841W-01/nixnet/rawframeformat/](url)
[http://zone.ni.com/reference/en-XX/help/372841W-01/nixnet/tdms/](url)
[https://www.ni.com/de-de/support/documentation/supplemental/07/tdms-file-format-internal-structure.html](url)
[https://www.ni.com/de-de/support/documentation/supplemental/13/embedded-network-data-in-tdms.html](url)

But i am by far not able to do that

Greets
ThetomekK

adamreeve, 2020-02-15:
Hi,

I'm not convinced it makes sense for npTDMS to support reading this type of data, it seems like this is a higher level format built on top of TDMS so a separate package that uses npTDMS would be more appropriate.

Regardless, I don't have any plans to support NI-XNET myself, so either way somebody else would need to implement this in npTDMS or as a separate package.

Cheers,
Adam

ThetomekK, 2020-02-15:
Hi Adam,

thanks for your reply and assessment on how to deal with this.
I'll inform you, if i've fond a way to deal with that kind of data.

Cheers
Thomas

closed 2020-02-18 by adamreeve