adamreeve/npTDMS #158: Thermocouple scaling processed as Thermistor

agenecov, 2020-03-02:
I am logging thermocouple data via DAQmx to TDMS. When I use this library to load the resulting data, I get "Exception: Cannot compute data for scale 8," coming from line 335 in scaling.py in version 0.19.0. I also get a warning in the log: "Unsupported scale type: Thermistor."

I know that I am configuring the device to read a thermocouple, not a thermistor, based on other testing and visual inspection of the code, so I am unsure what is happening.

adamreeve, 2020-03-03:
Hi, I'll need a bit more information to understand what's happening here. What are the properties of the channel? You can access them with the `properties` attribute of the channel object or see them by running `tdmsinfo --properties <path to TDMS file>`.

agenecov, 2020-03-03:
@adamreeve here is a printout for a single thermocouple channel:
name:
  /'ai'
    /'ai'/'t0'
      data type: DaqMxRawData
      properties:
      NI_Scaling_Status: unscaled
      NI_Number_Of_Scales: 14
      NI_Scale[1]_Scale_Type: Linear
      NI_Scale[1]_Linear_Slope: 9.31322630126633e-09
      NI_Scale[1]_Linear_Y_Intercept: 0.0
      NI_Scale[1]_Linear_Input_Source: 0
      NI_Scale[3]_Scale_Type: Linear
      NI_Scale[3]_Linear_Slope: 9.31322630126633e-09
      NI_Scale[3]_Linear_Y_Intercept: 0.0
      NI_Scale[3]_Linear_Input_Source: 2
      NI_Scale[4]_Scale_Type: Subtract
      NI_Scale[4]_Subtract_Left_Operand_Input_Source: 1
      NI_Scale[4]_Subtract_Right_Operand_Input_Source: 3
      NI_Scale[5]_Scale_Type: Linear
      NI_Scale[5]_Linear_Slope: 1000000.0
      NI_Scale[5]_Linear_Y_Intercept: -0.0
      NI_Scale[5]_Linear_Input_Source: 4
      NI_Scale[7]_Scale_Type: Linear
      NI_Scale[7]_Linear_Slope: 2.980232416405226e-07
      NI_Scale[7]_Linear_Y_Intercept: 0.0
      NI_Scale[7]_Linear_Input_Source: 6
      NI_Scale[8]_Scale_Type: Thermistor
      NI_Scale[8]_Thermistor_Resistance_Configuration: 3
      NI_Scale[8]_Thermistor_Excitation_Type: 10322
      NI_Scale[8]_Thermistor_Excitation_Value: 2.5
      NI_Scale[8]_Thermistor_R1_Reference_Resistance: 10000.0
      NI_Scale[8]_Thermistor_Lead_Wire_Resistance: 0.0
      NI_Scale[8]_Thermistor_A: 0.0012873851
      NI_Scale[8]_Thermistor_B: 0.00023575235
      NI_Scale[8]_Thermistor_C: 9.497806e-08
      NI_Scale[8]_Thermistor_Temperature_Offset: 1.0
      NI_Scale[8]_Thermistor_Input_Source: 7
      NI_Scale[9]_Scale_Type: Linear
      NI_Scale[9]_Linear_Slope: 1.0
      NI_Scale[9]_Linear_Y_Intercept: -273.15
      NI_Scale[9]_Linear_Input_Source: 8
      NI_Scale[10]_Scale_Type: Thermocouple
      NI_Scale[10]_Thermocouple_Thermocouple_Type: 10073
      NI_Scale[10]_Thermocouple_Scaling_Direction: 1
      NI_Scale[10]_Thermocouple_Input_Source: 9
      NI_Scale[11]_Scale_Type: Add
      NI_Scale[11]_Add_Left_Operand_Input_Source: 5
      NI_Scale[11]_Add_Right_Operand_Input_Source: 10
      NI_Scale[12]_Scale_Type: Thermocouple
      NI_Scale[12]_Thermocouple_Thermocouple_Type: 10073
      NI_Scale[12]_Thermocouple_Scaling_Direction: 0
      NI_Scale[12]_Thermocouple_Input_Source: 11
      NI_Scale[13]_Scale_Type: Linear
      NI_Scale[13]_Linear_Slope: 1.8
      NI_Scale[13]_Linear_Y_Intercept: 32.0
      NI_Scale[13]_Linear_Input_Source: 12
      NI_ChannelName: t0
      unit_string: F
      NI_UnitDescription: F
      wf_start_time: 2020-03-03T15:09:02.048792
      wf_increment: 0.013333325
      wf_start_offset: 0.0
      wf_samples: 1

agenecov, 2020-03-03:
@adamreeve oh, I think I might understand what's happening here. I'm specifying to use an internal cold junction compensator on the DAQ card, which is probably a thermistor. That's kind of a bummer. Is there a reason that nptdms doesn't support thermistor scaling?

adamreeve, 2020-03-03:
Ok yes that would make sense. Yes it should be fairly straightforward to add support for this. It's just a matter of figuring out how to compute the temperature value from the input value and the various properties associated with the scale (Resistance_Configuration, Excitation_Type, Excitation_Value, R1_Reference_Resistance, Lead_Wire_Resistance, A, B, C and Temperature_Offset), and making sure the units match what DAQmx uses.

It looks like the A, B and C parameters correspond to terms in the Steinhart-Hart equation (https://en.m.wikipedia.org/wiki/Steinhart%E2%80%93Hart_equation). The Excitation_Type corresponds to voltage excitation (see https://nidaqmx-python.readthedocs.io/en/latest/constants.html#nidaqmx.constants.ExcitationVoltageOrCurrent) but could also potentially be 10134 for current excitation. And Resistance_Configuration = 3 indicates a 3 wire circuit configuration.

agenecov, 2020-03-03:
@adamreeve yep that all looks correct to me. thank you so much for the quick response and being willing to implement!

adamreeve, 2020-03-03:
No problem, although I'm not sure how long it will take to find some time to implement this. If you could provide an example file that uses this scaling that would help verify the implementation is correct.

closed 2020-03-14 by adamreeve

adamreeve, 2020-03-14:
This has now been implemented in version 0.20.0

PR #162 (2020-03-14): Implement thermistor scaling
Fixes #158