adamreeve/npTDMS #182: Converting to a data frame in Spyder 4

gkm147, 2020-04-09:
Hi Adam, 

I recently installed Spyder 4.0.1 and I am having issues converting the data from a group into a dataframe. For example, I used the following code in Python 3.7 in Spyder 3 and it works perfectly. 

tdms_file=TdmsFile("example.tdms")
tdms_groups=tdms_file.groups()
scaled_data = tdms_file.object(tdms_groups[0]).as_dataframe()

However, in Spyder 4.0, it says 'TdmsGroup' object has no attribute 'replace'. I was wondering if you could help me with this issue.

zariiii9003, 2020-04-09:
I just wanted to create the same issue. Your code works with version 0.22.0 though.
The API change broke backwards compatibility.

adamreeve, 2020-04-09:
Hi, yes version 0.23.0 introduced a breaking change, you now need to use:

```python
tdms_file=TdmsFile("example.tdms")
tdms_groups=tdms_file.groups()
scaled_data = tdms_groups[0].as_dataframe()
```

The `groups` method now returns a list of `TdmsGroup` objects instead of group names. There's some more information on this in #161 and the documentation has been updated. I had tried to keep some backwards compatibility but missed that the group might be passed to the `object` method.

adamreeve, 2020-04-10:
I've released version 0.24.1 with a fix so that the `object` method now works when passed a `TdmsGroup`, but the `object` method will eventually be removed so you should either fix the version of npTDMS you're using to 0.22.0 or update your code for the recent API changes.

As discussed in #179, I haven't guaranteed backwards compatibility previously but plan on releasing a 1.0.0 version soon and then following semantic versioning.

gkm147, 2020-04-10:
Thank you Adam. I am currently using the version 0.23.0. I used the following code. 
```
tdms_file=TdmsFile("example.tdms")
tdms_groups=tdms_file.groups()
scaled_data = tdms_groups[0].as_dataframe()
```
I am unable to access the first groups data but could access the second one. . It says
```
 index -1 is out of bounds for axis 0 with size 0
```
I added an example file. 
[example.zip](https://github.com/adamreeve/npTDMS/files/4462874/example.zip)

adamreeve, 2020-04-10:
Looks like this is unrelated to the recent API changes as this also fails on 0.22.0. The problem here is that there are channels with polynomial scaling applied but with no coefficients, which isn't handled correctly. Eg this channel only has polynomial scaling:
```
    /'Data (Scaled)'/'RSG3_TEMP'
      data type: DoubleFloat
      length: 60000
      properties:
        NI_UnitDescription: C
        unit_string: C
        Signal_Type: AI
        NI_Scaling_Status: unscaled
        NI_Number_Of_Scales: 1
        NI_Scale[0]_Scale_Type: Polynomial
        NI_Scale[0]_Polynomial_Coefficients_Size: 0
        NI_Scale[0]_Polynomial_Input_Source: 4294967295
        NI_ArrayColumn: 17
```
and this has a linear and then polynomial scaling applied:
```
    /'Data (Scaled)'/'Geokon1'
      data type: DoubleFloat
      length: 60000
      properties:
        NI_UnitDescription:
        unit_string:
        Signal_Type: RSG
        NI_Scaling_Status: unscaled
        NI_Number_Of_Scales: 2
        NI_Scale[0]_Scale_Type: Linear
        NI_Scale[0]_Linear_Slope: 2.5
        NI_Scale[0]_Linear_Y_Intercept: 0.0
        NI_Scale[0]_Linear_Input_Source: 4294967295
        NI_Scale[1]_Scale_Type: Polynomial
        NI_Scale[1]_Polynomial_Coefficients_Size: 0
        NI_Scale[1]_Polynomial_Input_Source: 0
        NI_ArrayColumn: 18
```
It seems like the correct way to handle this is to return an array of zeros for these channels.

adamreeve, 2020-04-10:
This should be fixed in version 0.24.2

closed 2020-04-15 by adamreeve