adamreeve/npTDMS #185: Reading all the metadata

gkm147, 2020-04-16:
Adam, 

I am trying to read in the metadata for a group using the following code. 
```
tdms_file=TdmsFile.read(example.tdms")
tdms_groups=tdms_file.groups()
Group_Properties=tdms_groups[0].properties
```
However, it is not reading in all the metadata. For example, if you open the nptdms file as an Excel file, you can see its only reading in the data from the columns with italicized headers and not from the columns with the bold headers . Can you help me figure out what is going on here. The example file is attached. 

[Example.zip](https://github.com/adamreeve/npTDMS/files/4488379/Example.zip)

adamreeve, 2020-04-16:
Hi, I don't have Excel but have the Open Office importer which I would guess behaves the same. It looks to me like the bold columns are only "Description" and "Channels". The other properties are "Session", "Test_Section", "Configuration_Id", "wf_start_time" and "wf_increment" which all look like they're working correctly. The "Description" property is empty so it doesn't look like anything is really missing there, and then the "Channels" property isn't a normal property but just looks to be the number of channels in the group, which you could get with `len(tdms_groups[0])` or `len(tdms_groups[0].channels())`

gkm147, 2020-04-17:
Thank you. I figured out that the columns in the bold are the defaults NI pproperties. I was just wondering if I want the default properties of the channels (such as datatype, Unit, Length, etc. ) do I have to get them manually or is there a way nptdms could get them in automatically?

adamreeve, 2020-04-18:
You would need to get them manually, I don't think there's much need for npTDMS to add these as additional properties automatically.

If a channel has a unit it will have `unit_string` and `NI_UnitDescription` properties. Length can be retrieved with `len(channel)`, and for the data type you can use the `dtype` attribute which returns a numpy dtype.

closed 2020-04-24 by adamreeve