adamreeve/npTDMS #186: [Feature Request] `pathlib.Path` compatibility of index_file_path

alexandreCameron, 2020-04-27:
## Proposal

Current use of path is well suited for `str` but not for ``pathlib.Path`

For instance:

```python
index_file_path = tdms_file + '_index'
```

could be adapted to:

```python
index_file_path = Path(Path(tdms_file).parent, Path(tdms_file).name + '_index')
```

I would be happy to suggest a PR on that issue

## Current undesired behavior

```python
tdms_file = PosixPath('/usr/src/app/data/my_tdms.tdms')

    def __init__(self, tdms_file):
        """ Initialise a new TdmsReader
    
        :param tdms_file: Either the path to the tdms file to read or an already
            opened file.
        """
        self._segments = None
        self._prev_segment_objects = {}
        self.object_metadata = OrderedDict()
        self._file_path = None
        self._index_file_path = None
    
        self._segment_channel_offsets = None
        self._segment_chunk_sizes = None
    
        if hasattr(tdms_file, "read"):
            # Is a file
            self._file = tdms_file
        else:
            # Is path to a file
            self._file = open(tdms_file, 'rb')
            self._file_path = tdms_file
>           index_file_path = tdms_file + '_index'
E           TypeError: unsupported operand type(s) for +: 'PosixPath' and 'str'
```

## Reference

https://docs.python.org/3/library/pathlib.html

adamreeve, 2020-04-27:
Hi, thanks for the bug report. Yes this would be good to support but I'm still maintaining support for Python 2.7 for now so it would be good if this could also work on 2.7 without needing to install the pathlib package from PyPI.

adamreeve, 2020-04-28:
I've fixed this on master and will make a bugfix release soon.

PR #187 (2020-04-28): Fix for passing file paths as pathlib.Path
Fixes #186
closed 2020-04-28 by adamreeve