adamreeve/npTDMS #188: tdms_group.to_dataframe() TypeError

mike-lloyd03, 2020-04-28:
I switched my Python installation from Python 3.8 (32-bit) to Anaconda which uses Python 3.7.6 (64-bit). Now I am no longer able to use .to_dataframe() when reading from a TDMS file. I get the following error
`TypeError: a bytes-like object is required, not 'float'`

Preview of code:
```python
import pandas as pd
import numpy as np
from pandas import ExcelWriter
from pandas import ExcelFile
from nptdms import TdmsFile
import sys
import matplotlib.pyplot as plt

def filter_and_interpolate_data (path_to_file):
  tdms_file_name = path_to_file
  tdms_file = TdmsFile(tdms_file_name)
  output_file = f'./Filtered Data/{tdms_file_name[0:-5]} Filtered.xlsx'
  data_blocks = ((0, 25), (490, 500), (990, 1000))

  # Create a list of dataframes from each TDMS group
  tdms_data = [group.as_dataframe() for group in tdms_file.groups()]

  ...
```

Full file:
[filterTDMS.zip](https://github.com/adamreeve/npTDMS/files/4546899/filterTDMS.zip)

adamreeve, 2020-04-28:
Hi, could you provide the full stack trace of the error, or an example TDMS file that triggers this issue? For simple test cases this does work on 64 bit Python 3.7 so there must be something different about your files that triggers this problem.

mike-lloyd03, 2020-04-28:
Here's the TDMS I was using:
[1309 027 IIG2 Sample 2 DATA.zip](https://github.com/adamreeve/npTDMS/files/4548477/1309.027.IIG2.Sample.2.DATA.zip)

Here's the traceback:

```
Traceback (most recent call last):
  File "c:/Users/mlloyd/Desktop/Local Projects/TR1309 Stick Slip/filterTDMS.py", line 64, in <module>
    filter_and_interpolate_data('1309 027 IIG2 Sample 2 DATA.tdms')
  File "c:/Users/mlloyd/Desktop/Local Projects/TR1309 Stick Slip/filterTDMS.py", line 16, in filter_and_interpolate_data
    tdms_data = [group.as_dataframe() for group in tdms_file.groups()]
  File "c:/Users/mlloyd/Desktop/Local Projects/TR1309 Stick Slip/filterTDMS.py", line 16, in <listcomp>
    tdms_data = [group.as_dataframe() for group in tdms_file.groups()]
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\nptdms\tdms.py", line 478, in as_dataframe
    return pandas_export.from_group(self, time_index, absolute_time, scaled_data)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\nptdms\export\pandas_export.py", line 40, in from_group
    return _channels_to_dataframe(channels_to_export, time_index, absolute_time, scaled_data)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\nptdms\export\pandas_export.py", line 77, in _channels_to_dataframe
    return pd.DataFrame.from_dict(dataframe_dict)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\frame.py", line 1247, in from_dict
    return cls(data, index=index, columns=columns, dtype=dtype)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\frame.py", line 435, in __init__
    mgr = init_dict(data, index, columns, dtype=dtype)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\internals\construction.py", line 254, in init_dict
    return arrays_to_mgr(arrays, data_names, index, columns, dtype=dtype)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\internals\construction.py", line 69, in arrays_to_mgr
    arrays = _homogenize(arrays, index, dtype)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\internals\construction.py", line 311, in _homogenize
    val = val.reindex(index, copy=False)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\series.py", line 4030, in reindex
    return super().reindex(index=index, **kwargs)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\generic.py", line 4544, in reindex
    axes, level, limit, tolerance, method, fill_value, copy
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\generic.py", line 4567, in _reindex_axes
    allow_dups=False,
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\generic.py", line 4613, in _reindex_with_indexers
    copy=copy,
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\internals\managers.py", line 1257, in reindex_indexer
    new_blocks = self._slice_take_blocks_ax0(indexer, fill_tuple=(fill_value,))
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\internals\managers.py", line 1306, in _slice_take_blocks_ax0
    fill_tuple=fill_tuple,
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\internals\blocks.py", line 1291, in take_nd
    values, indexer, axis=axis, allow_fill=allow_fill, fill_value=fill_value
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\algorithms.py", line 1662, in take_nd
    func(arr, indexer, out, fill_value)
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\algorithms.py", line 1466, in func2
    arr, indexer, out, axis=axis, fill_value=fill_value, mask_info=mask_info
  File "C:\Users\mlloyd\Anaconda3\lib\site-packages\pandas\core\algorithms.py", line 1332, in _take_nd_object
    out[tuple(outindexer)] = fill_value
TypeError: a bytes-like object is required, not 'float'
```

adamreeve, 2020-04-28:
Thanks, it looks like this is caused by the empty `Dynamic Data/Temp` channel, which Pandas doesn't like. I'll try to get a fix in for this soon.

mike-lloyd03, 2020-04-28:
I just noticed that the Anaconda package is using Pandas 1.0.1 but the latest version is 1.0.3. Conda says Anaconda requires 1.0.1 so that could be the problem. I'll try updating it anyway and see if that fixes it.

mike-lloyd03, 2020-04-28:
Looks like Anaconda won't let me update pandas manually. I'll have to play more with this tomorrow.

closed 2020-04-29 by adamreeve

adamreeve, 2020-04-29:
This should be fixed now if you upgrade npTDMS to version 0.25.1

mike-lloyd03, 2020-04-29:
Worked. Thanks so much!

PR #189 (2020-04-29): Handle void arrays when converting to Pandas dataframe 
Fixes #188