adamreeve/npTDMS #190: Out of bounds nanosecond timestamp

tzonghao, 2020-04-29:
Since version 0.25, I get an error parsing a timestamp with Pandas when there's an TDMS index file, like this:

`pandas._libs.tslibs.np_datetime.OutOfBoundsDatetime: Out of bounds nanosecond timestamp: 7802-09-26 06:38:33`

If I remove the index file, things go back to normal. Any idea what could cause this error?

adamreeve, 2020-04-29:
Hi, that seems odd, it sounds like the data might be read at the wrong location. Are you able to provide an example tdms and tdms_index file with this problem?

tzonghao, 2020-04-30:
Here's two example files (link expires in 7 days):
https://send.firefox.com/download/874e399fa4b73f88/#FYe0kZHyJrXGDCuWsZpKHg

Thanks for looking into this.

adamreeve, 2020-04-30:
So it looks like the index file has come from a different tdms file, which is what's causing the data to all be read incorrectly. Eg. when I read the file without the index it has a property `tdms.properties['name']` which is `Aligned Data2013-10-07 All Day` but in the index this property is `2013-10-07 All Day`. A lot of the channels in the index file have length of 1440 but in the data file the lengths are actually 1439.

I could probably add some checks that verify the segment start positions match up with what the index says which should catch this sort of problem.

tzonghao, 2020-05-01:
That'd be a good solution. Someone must have changed the TDMS file while leaving the old index file in place. For now, I can simply remove those outdated index files. Again, thanks for checking.

adamreeve, 2020-05-03:
I've created #195 to implement a check that the data file matches the index file, so I'll close this issue.

closed 2020-05-03 by adamreeve