adamreeve/npTDMS #198: Preserving raw LabVIEW timestamp

Achilles1515, 2020-05-07:
I have an existing TDMS --> H5 conversion process (in .NET) that I am porting to Python, mainly because of the DAQmx support in this library. One issue I am running into involves the LabVIEW timestamp.

Currently, we directly copy the LabVIEW timestamp (two 64-bit ints) as a dataset in the H5 file.

 It looks like this library is converting the timestamp into a NumPy `datetime64`, hence the limitation stated in the README:

>TDMS files support timestamps with a resolution of 2^-64 seconds but these are read as numpy datetime64 values with microsecond resolution.

This is fine and all for convenience, but I am looking for access to the actual timestamp integers.

1) Is there a way to access the original LabVIEW timestamp integers using this library?
2) If not, can support be added to obtain these original values for a channel? (possibly as attributes on the `TdmsChannel` class or wherever you see fit).

Again, losing precision to conveniently work with a datetime object is a fine feature, but I think the user should at least still be able to access the original values in full precision.

adamreeve, 2020-05-07:
Hi. There isn't currently any way to access the raw timestamps. When reading the timestamps the raw integers are discarded after converting to a numpy datetime.

It would make sense to expose these raw integers though, I'll have to think about how best to do this.

Achilles1515, 2020-05-08:
I appreciate you looking into it. Thanks for your work on the library.

Achilles1515, 2020-05-15:
In case anyone is curious -
As a workaround for the time being, I have modified the `TimeStamp` class in the "types.py" file to return a dictionary containing the raw LabVIEW timestamp integers, as well as the NumPy `datetime64` value.

```Python
class TimeStamp(TdmsType):
    #...
    @classmethod
    def read(cls, file, endianness="<"):
        data = file.read(16)
        if endianness == "<":
            (second_fractions, seconds) = _struct_unpack(
                endianness + 'Qq', data)
        else:
            (seconds, second_fractions) = _struct_unpack(
                 endianness + 'qQ', data)
        micro_seconds = int(
            float(second_fractions) / cls._fractions_per_microsecond)

        # CHANGE: Preserve raw timestamp values
        converted_time = (cls._tdms_epoch + np.timedelta64(seconds, 's') +
                          np.timedelta64(micro_seconds, 'us'))
        return {"raw": [seconds, second_fractions], "converted": converted_time}
    #...
```

adamreeve, 2020-05-15:
This is now implemented on the master branch (see #200) and documented at https://nptdms.readthedocs.io/en/latest/reading.html#timestamps

closed 2020-05-15 by adamreeve