adamreeve/npTDMS #199: TdmsFile.read() leaks memory

Achilles1515, 2020-05-12:
Something seems fishy with memory usage of the `TdmsFile.read()` method. It appears to do as it states - load the entire TDMS file in memory - but the memory doesn't appear to be released (at least in any sort of timely fashion) afterward.

Take this simple example:

```python
from nptdms import TdmsFile

count = 0
path = "...path to some 66 MB TDMS file"
while count < 10:
    tdms_file = TdmsFile.read(path)
    tdms_file.close()	# pretty sure this does nothing here
    count += 1

time.sleep(10)
```

After the loop has complete, memory usage of the Python script is **460 MB**.
Now, manually clearing the `channel._raw_data.data` value will net the [more] expected memory usage of the script:

```python
from nptdms import TdmsFile

count = 0
path = "...path to some 66 MB TDMS file"
while count < 10:
    tdms_file = TdmsFile.read(path)
    for group in tdms_file.groups():
        for channel in group.channels():
            channel._raw_data.data = None  # <-- HERE
    tdms_file.close()
    count += 1

time.sleep(10)
```

Memory usage at end of script = **17 MB**

Though, simply accessing the channel data increases memory usage back up to 460 MB (the data is kept in cached properties?). 
```python
from nptdms import TdmsFile

count = 0
path = "...path to some 66 MB TDMS file"
while count < 10:
    tdms_file = TdmsFile.read(path)
    for group in tdms_file.groups():
        for channel in group.channels():
            z = channel[:]  # <-- HERE
            channel._raw_data.data = None
    tdms_file.close()
    count += 1

time.sleep(10)
```

Manually calling `gc.collect()` immediately after the `while` loop does free some memory, but the script memory usage is still ~80 MB afterwards.

How are we supposed to be freeing memory here using `TdmsFile.read()`? In all these examples, the `tdms_file` variable is scoped inside the `while` loop, yet the `tdms_file` memory is not being freed once exited or between loop iterations. My Python scripts are consistently running out of memory whenever I try to use `TdmsFile.read()` while looping over a number of TDMS files of decent size.

adamreeve, 2020-05-13:
There are some circular references between `TdmsFile` and `TdmsChannel` which prevent Python from immediately freeing memory when a `TdmsFile` goes out of scope. Using weakrefs here might help but could also break some code. I think it should be possible to refactor to fix this and remove the circular references.

Until then calling `gc.collect` would be the best option if you need to free memory immediately.

> Manually calling gc.collect() immediately after the while loop does free some memory, but the script memory usage is still ~80 MB afterwards (one TDMS file data, I suppose).

80 MB sounds about right. As you say, calling close doesn't have any effect when using `TdmsFile.read`, that's there to close file handles when using `TdmsFile.open`, so there will still be the memory used by one file.

> Though, simply accessing the channel data increases memory usage back up to 460 MB (the data is kept in cached properties?).

Yes, accessing the data causes it to be cached internally, you'd also need to set `_cached_prop_data` to None if data has been accessed. This is because sometimes scaling is applied to the raw data, and the scaled data is then cached.

closed 2020-05-19 by adamreeve

Achilles1515, 2020-05-19:
@adamreeve Thanks for the fix. Memory does appear to be freed as normal.

Though with these new changes, the following code now throws the following exception:
`AttributeError: 'NoneType' object has no attribute 'close'`

```python
tdms_file = TdmsFile.read(path)
tdms_file.close()
```

As previously mentioned, the `tdms_file.close()` doesn't really do anything when `TdmsFile.read()`. But this is technically a breaking change since if previously didn't throw an exception.

The problem is in the following code:

```python
class TdmsReader(object):
    # ...

    def close(self):
        if self._file_path is not None:
            # File path was provided so we opened the file and
            # should close it.
            self._file.close()
        # Otherwise always remove reference to the file
        self._file = None
```  

With the snippet I provided, `self._file_path` is not `None`, but `self._file` is `None`.
Can another check be added to ensure `self._file` is not `None`?

adamreeve, 2020-05-19:
Ah sorry about that, yes it wasn't intentional to change that behaviour, I've pushed a fix for this now.

PR #201 (2020-05-19): Fix circular reference between TdmsFile and TdmsChannel
Fixes #199. This circular reference meant that when a TdmsFile went out of scope the memory it used wasn't immediately freed but was only freed after a GC collection.
commit 4a5c9df2 (2020-05-19): Fix circular reference between TdmsFile and TdmsChannel (#201)

Fixes #199. This circular reference meant that when a TdmsFile went out of scope the memory it used wasn't immediately freed but was only freed after a GC collection.