adamreeve/npTDMS #206: npTDMS cannot read what it wrote - unicode string related?

bolu-atx, 2020-07-22:
Hi Adam, thanks for open sourcing this library and providing a very useful feature, I have a quick question...

I wrote a TDMS file exporting data from a multi-column numpy array following the example in the documentation as follows:

```python
from nptdms import TdmsWriter, RootObject, GroupObject, ChannelObject
root_object = RootObject(properties=header_data)

group_object = GroupObject("Currents")
group_object_voltage = GroupObject("Voltage")

with TdmsWriter("output.tdms") as tdms_writer:    
    tdms_writer.write_segment([
            root_object,
            group_object])
    for channel_index in range(num_channels-1):
        channel_number = channel_index+1
        current_data  = data[:, channel_index]
        channel_object = ChannelObject("Currents", "channel_{}".format(channel_number), data = current_data)
        tdms_writer.write_segment([channel_object])

    voltage_data = data[:, -1]
    voltage_channel = ChannelObject("Voltage", "voltage", data)
    tdms_writer.write_segment([group_object_voltage, voltage_channel])
```

However, when I try to read it:
```
from nptdms import TdmsFile

with TdmsFile.open("output.tdms") as tdms_file:
    # Use tdms_file
    tdms_file.groups()
```

I get an error
```
~/miniconda2/envs/py37/lib/python3.8/site-packages/nptdms/reader.py in _read_lead_in(self, file, segment_position, is_index_file)
    242             raise EOFError
    243         if tag != expected_tag:
--> 244             raise ValueError(
    245                 "Segment does not start with %r, but with %r" % (expected_tag, tag))
    246 

ValueError: Segment does not start with b'TDSm', but with b'\x8c"\x04>'
```

adamreeve, 2020-07-22:
Hi, thanks for the bug report, is it possible to provide a complete working example that reproduces the issue? In this example I can't see what `data`, `header_data` and `num_channels` are.

bolu-atx, 2020-07-22:
Hi Adam,

I've made a small test program that reproduces this issue. I was able to reproduce it using Python 3.8 and Python 2.7.

```python
import numpy as np

data = np.random.randn(100,5).astype(np.float32)

num_channels = 5

from nptdms import TdmsWriter, RootObject, GroupObject, ChannelObject
root_object = RootObject()
group_object = GroupObject("Currents")
group_object_voltage = GroupObject("Voltage")

with TdmsWriter("output.tdms") as tdms_writer:    
    tdms_writer.write_segment([
            root_object,
            group_object])
    for channel_index in range(num_channels-1):
        channel_number = channel_index+1
        current_data  = data[:, channel_index]
        channel_object = ChannelObject("Currents", "channel_{}".format(channel_number), data = current_data)
        tdms_writer.write_segment([channel_object])

    voltage_data = data[:, -1]
    voltage_channel = ChannelObject("Voltage", "voltage", data)
    tdms_writer.write_segment([group_object_voltage, voltage_channel])

from nptdms import TdmsFile

with TdmsFile.open("output.tdms") as tdms_file:
    # Use tdms_file
    tdms_file.groups()
```

bolu-atx, 2020-07-22:
full stacktrace of the error:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-33-366d7f5417d9> in <module>
      1 from nptdms import TdmsFile
      2 
----> 3 with TdmsFile.open("output.tdms") as tdms_file:
      4     # Use tdms_file
      5     tdms_file.groups()

~/miniconda2/envs/py37/lib/python3.8/site-packages/nptdms/tdms.py in open(file, raw_timestamps, memmap_dir)
     91             temporary file directory.
     92         """
---> 93         return TdmsFile(
     94             file, raw_timestamps=raw_timestamps, memmap_dir=memmap_dir, read_metadata_only=True, keep_open=True)
     95 

~/miniconda2/envs/py37/lib/python3.8/site-packages/nptdms/tdms.py in __init__(self, file, raw_timestamps, memmap_dir, read_metadata_only, keep_open)
    134         self._reader = TdmsReader(file)
    135         try:
--> 136             self._read_file(self._reader, read_metadata_only)
    137         finally:
    138             if not keep_open:

~/miniconda2/envs/py37/lib/python3.8/site-packages/nptdms/tdms.py in _read_file(self, tdms_reader, read_metadata_only)
    230 
    231     def _read_file(self, tdms_reader, read_metadata_only):
--> 232         tdms_reader.read_metadata()
    233 
    234         # Use object metadata to build group and channel objects

~/miniconda2/envs/py37/lib/python3.8/site-packages/nptdms/reader.py in read_metadata(self)
     83                     start_position = file.tell()
     84                     try:
---> 85                         segment = self._read_segment_metadata(
     86                             file, segment_position, previous_segment, reading_index_file)
     87                     except EOFError:

~/miniconda2/envs/py37/lib/python3.8/site-packages/nptdms/reader.py in _read_segment_metadata(self, file, segment_position, previous_segment, is_index_file)
    220             self, file, segment_position, previous_segment=None, is_index_file=False):
    221         (position, toc_mask, endianness, data_position, raw_data_offset,
--> 222          next_segment_offset, next_segment_pos) = self._read_lead_in(file, segment_position, is_index_file)
    223 
    224         segment_args = (

~/miniconda2/envs/py37/lib/python3.8/site-packages/nptdms/reader.py in _read_lead_in(self, file, segment_position, is_index_file)
    242             raise EOFError
    243         if tag != expected_tag:
--> 244             raise ValueError(
    245                 "Segment does not start with %r, but with %r" % (expected_tag, tag))
    246 

ValueError: Segment does not start with b'TDSm', but with b'\x8c"\x04>'

```

adamreeve, 2020-07-23:
It looks like the problem is this line in your code where you pass the full 2d array as the voltage channel data:
```python
voltage_channel = ChannelObject("Voltage", "voltage", data)
```
If I change this to use the voltage data then it works:
```python
voltage_channel = ChannelObject("Voltage", "voltage", voltage_data)
```

The `ChannelObject` class assumes the data is 1d but doesn't actually check this, so I'll fix this so that it raises a more helpful error when writing.

bolu-atx, 2020-07-23:
I see. Thanks, I should've used `flatten()`. Thanks

closed 2020-07-24 by bolu-atx