adamreeve/npTDMS #208: Thermocouple Scaling very slow

rddaz2013, 2020-07-28:
Hi,

within a juptyer notebook it needs a lot of time if i want to plot a temperature. 
50.000 datapoints with non-temperature scaleing do not show such a time requirement.

I think the speed is lost in the https://pypi.org/project/thermocouples_reference/
I will prepare a notebook and data for testing in github.

rddaz2013, 2020-07-28:
I have looked into the tdms file..... thermocouples have two Scaling Part's in the Labview TDMS Explorer (one is polynomial and say only what Thermocouple is uses).

It is possible to use only the first scaling without the need for the thermocouples_reference modul.
I think this modul is not fast within numpy arrays. 

René

adamreeve, 2020-07-28:
I doubt you can ignore the thermocouple scaling and only use the polynomial scaling, quite often data has multiple scalings that must all be applied.

The thermocouples_reference documentation says:
> Lookup tables also often also include approximate polynomials for temperature lookup based on a given compensated emf value. Such inverse polynomials are not included in this module; rather, the inverse lookup is based on numerically searching for a solution on the exact emf function.

So it's not surprising this is slow. There are inverse polynomial coefficients on the NIST website (eg. https://srdata.nist.gov/its90/type_k/kcoefficients_inverse.html) that could be used instead but the result would be slightly less accurate. There's a Python script on GitHub at https://github.com/andygock/Thermocouple that implements these inverse lookup tables but it isn't available as a library on PyPI.

I think it would be worth testing using the inverse lookup polynomials and see how this compares to LabView's output.

rddaz2013, 2020-07-28:
Thx

> and only use the polynomial scaling

how to do his with npTDMS - because labview delivers the inverse lookup polynomials.

adamreeve, 2020-07-28:
I'm not really following what you're saying, are you able to provide an example file or at least show the properties dictionary of the channel that contains the scaling information?

rddaz2013, 2020-07-29:
So there is the testdata - https://github.com/rddaz2013/npTDMS/blob/master/datatest/Test_Data.ipynb

```
OrderedDict([('NI_Scaling_Status', 'unscaled'),
             ('NI_Number_Of_Scales', 3),
             ('NI_Scale[1]_Scale_Type', 'Polynomial'),
             ('NI_Scale[1]_Polynomial_Coefficients_Size', 4),
             ('NI_Scale[1]_Polynomial_Coefficients[0]', 3717.2217387015417),
             ('NI_Scale[1]_Polynomial_Coefficients[1]', 6.585674458677389),
             ('NI_Scale[1]_Polynomial_Coefficients[2]', 1.427913774635689e-09),
             ('NI_Scale[1]_Polynomial_Coefficients[3]',
              1.0763380172505354e-13),
             ('NI_Scale[1]_Polynomial_Input_Source', 0),
             ('NI_Scale[2]_Scale_Type', 'Thermocouple'),
             ('NI_Scale[2]_Thermocouple_Thermocouple_Type', 10073),
             ('NI_Scale[2]_Thermocouple_Scaling_Direction', 0),
             ('NI_Scale[2]_Thermocouple_Input_Source', 1),
             ('NI_ChannelName', 'Dev1/ai0'),
             ('unit_string', 'Deg C'),
             ('NI_UnitDescription', 'Deg C'),
             ('wf_start_time', numpy.datetime64('2020-07-22T13:51:03.835663')),
             ('wf_increment', 0.001),
             ('wf_start_offset', 0.0),
             ('wf_samples', 1)])
```

npTDMS uses "NI_Scale[2]_Scale_Type', 'Thermocouple" - how can i use the 'NI_Scale[1]_Scale_Type', 'Polynomial' that must be the inverse lookup polynomials.

adamreeve, 2020-07-30:
I don't think that polynomial scaling is the inverse lookup. I'm not sure what it is, but if you look at the `Input_Source` properties you can see that the input for the polynomial is scale 0, which is the raw data, then the input for the thermocouple scale is 1, which is the polynomial scale, so to get the fully scaled data first the polynomial scaling is applied, then the output of that is scaled using the thermocouple scaling. npTDMS applies both these scales in turn and just using the polynomial scaling won't give you a sensible result.

adamreeve, 2020-08-29:
I've started implementing a replacement for the thermocouple scaling code in the [thermocouple_scaling branch](https://github.com/adamreeve/npTDMS/tree/thermocouple_scaling). It uses the inverse polynomials from the NIST website rather than searching for a solution in the forwards polynomials, and is around 500 times faster to calculate a voltage from a temperature and 5,000 times faster to calculate a temperature from a voltage in my benchmarks. The results are also closer to what the TDMS importer gives, so that is probably also using the inverse polynomials.

PR #215 (2020-09-02): New thermocouple scaling implementation
Replaces use of the thermocouples_reference module with our own implementation of the NIST reference functions. Uses the approximate inverse functions for better performance and compatibility with LabVIEW rather than searching for a solution in the forward functions (as thermocouples_reference does).

Fixes #208 

Looking into the thermocouples_reference module I realised I could improve performance a lot when converting from temperature to voltage as npTDMS currently converts one value at a time to so that it can catch `ValueError`s when an input is out [... cut]
closed 2020-09-02 by adamreeve