adamreeve/npTDMS #216: Creating new ChannelObject from existing

227e, 2020-09-03:
I have the requirement to take a tdms file, apply calibration data, and spit out a new file with the same properties, groups and channel names as the original. To do this, I create a new ChannelObject using the group and name of the existing file, fettle the data and add it to a list of replacement objects. Unfortunately you've deprecated the group property, so I had to use _path.group in the ChannelObject constructor to side step the deprecation warning. Can you suggest how I can best achieve this, or add the facility to create a new ChannelObject from an existing one.

adamreeve, 2020-09-05:
Hi, to explain the deprecation, the property was misnamed as `group` implies it returns the actual group object but it only returned the group's name. I was thinking I could re-add a group property later that returned the group object but I'm concerned that will lead to a circular reference which doesn't play nicely with Python memory management. So I think it probably makes sense to add a `group_name` attribute to the channel instead.

Having said that, surely you had to access the channel via a group object, so you should be able to use `group.name` to get the group's name?

227e, 2020-09-10:
Many thanks for your reply. Your deprecation case is a good one and I understand it fully. In my case I was using an earlier version of your library, and it allowed me to create a new channel from an existing one (by copying the path and properties). I appreciate that to select the channel I needed to know the group, but at the point I'm doing the replacement I only have a list of channels. I would be very grateful if you could enable or support a copy constructor for the tdms filter case. Either/or allow the data array of an existing ChannelObject to be unlinked or substituted with a new array. Many thanks

adamreeve, 2020-09-20:
I've added a new `group_name` property to `TdmsChannel` in #217 which should hopefully fix your use case, and this is included in version 1.0.0.

closed 2021-02-07 by adamreeve