adamreeve/npTDMS #218: Significant difference in loading speed

Gang-Dong, 2020-10-09:
Dear adamreeve:
  Thank you very much for reading my issue.
  We collected a set of data from an existing NI supporting instrument, with only one group and one channel, but its loading speed was very slow. The speed of reading the tdms file obtained by other instruments with the same data point using nptdms is normal.
  After checking, I don't know where the problem is. After reading the abnormal tdms file and then writing the new tdms file using a vi widget written in LabView, the reading speed using nptdms becomes very fast.
  I have uploaded the problematic tdms file and the re-read and saved tdms file (using labview) as attachments. I wonder if you can check the problem.

Thank you very much.

------------------------------------------------------------------------------
TMDS.zip:
    ----OPE2-22020-8-6 10-45-39-2.tdms   (Original file)
    ----OPE2-22020-8-6 10-45-39-2-new.tdms   (After conversion)

  
[TDMS.zip](https://github.com/adamreeve/npTDMS/files/5351874/TDMS.zip)

adamreeve, 2020-10-11:
Hi, the problem is that the original file has the segment data length set to 1. So even though the file stores the data as 5000002 contiguous double values, they're read one chunk (ie. one double) at a time which is very slow.

Running `tdmsinfo --debug --properties` for the original file gives:
```
[nptdms.reader DEBUG] Reading segment at 0
[nptdms.reader DEBUG] Property kTocMetaData is True
[nptdms.reader DEBUG] Property kTocRawData is False
[nptdms.reader DEBUG] Property kTocDAQmxRawData is False
[nptdms.reader DEBUG] Property kTocInterleavedData is False
[nptdms.reader DEBUG] Property kTocBigEndian is False
[nptdms.reader DEBUG] Property kTocNewObjList is True
[nptdms.reader DEBUG] Next segment offset = 58, raw data offset = 58
[nptdms.reader DEBUG] Data size = 0 b
[nptdms.base_segment DEBUG] Reading segment object metadata at 28
[nptdms.base_segment DEBUG] Reading metadata for object / with index header 0xffffffff
[nptdms.base_segment DEBUG] Reading 1 properties
[nptdms.base_segment DEBUG] Property 'name' = 'OPE2-22020-8-6 10-45-39-2'
[nptdms.reader DEBUG] Reading segment at 86
[nptdms.reader DEBUG] Property kTocMetaData is True
[nptdms.reader DEBUG] Property kTocRawData is True
[nptdms.reader DEBUG] Property kTocDAQmxRawData is False
[nptdms.reader DEBUG] Property kTocInterleavedData is True
[nptdms.reader DEBUG] Property kTocBigEndian is False
[nptdms.reader DEBUG] Property kTocNewObjList is True
[nptdms.reader DEBUG] Next segment offset = 152, raw data offset = 144
[nptdms.reader DEBUG] Data size = 8 b
[nptdms.base_segment DEBUG] Reading segment object metadata at 114
[nptdms.base_segment DEBUG] Reading metadata for object /'OPE2-22020/8/6 10:45:39' with index header 0xffffffff
[nptdms.base_segment DEBUG] Reading metadata for object /'OPE2-22020/8/6 10:45:39'/'RT-IO:AI0 Volt. [V]' with index header 0x00000014
[nptdms.tdms_segment DEBUG] Object data type: DoubleFloat
[nptdms.tdms_segment DEBUG] Object number of values in segment: 1
[nptdms.base_segment DEBUG] Reading 1 properties
[nptdms.base_segment DEBUG] Property 'NI_ArrayColumn' = 0
[nptdms.reader DEBUG] Reading segment at 266
[nptdms.reader DEBUG] Property kTocMetaData is True
[nptdms.reader DEBUG] Property kTocRawData is True
[nptdms.reader DEBUG] Property kTocDAQmxRawData is False
[nptdms.reader DEBUG] Property kTocInterleavedData is True
[nptdms.reader DEBUG] Property kTocBigEndian is False
[nptdms.reader DEBUG] Property kTocNewObjList is True
[nptdms.reader DEBUG] Next segment offset = 40000072, raw data offset = 64
[nptdms.reader DEBUG] Data size = 40000008 b
[nptdms.base_segment DEBUG] Reading segment object metadata at 294
[nptdms.base_segment DEBUG] Reading metadata for object /'OPE2-22020/8/6 10:45:39'/'RT-IO:AI0 Volt. [V]' with index header 0x00000000
[nptdms.reader INFO] Read metadata: Took 3.787600000000002 ms
/
  properties:
    name: OPE2-22020-8-6 10-45-39-2
  /'OPE2-22020/8/6 10:45:39'
    /'OPE2-22020/8/6 10:45:39'/'RT-IO:AI0 Volt. [V]'
      data type: DoubleFloat
      length: 5000002
      properties:
        NI_ArrayColumn: 0
```

This line shows the segment chunk size: `[nptdms.tdms_segment DEBUG] Object number of values in segment: 1`

It should be possible to handle this scenario better and read all chunks in a segment at once when data is interleaved.

adamreeve, 2020-10-11:
This should be fixed now if you upgrade to version 1.1.0

closed 2020-10-11 by adamreeve

Gang-Dong, 2020-10-13:
Thank you! I'll try it as soon as possible

Gang-Dong, 2020-10-13:
The problem has been solved. Thank you very much